AlbrechtL/welle.io #587: BBC National re-config - Frankenstein stream

andimik, 2020-07-29:
BBC National regularly uses re-configuration of the mux. It has been described first in https://www.rundfunkforum.de/viewtopic.php?f=11&t=55891&start=3330 (and following English language pages).

The issue is called Frankenstein stream due to a very strange output message (see below).

It's impossible to listen to the news (because the re-config is done exactly at the beginning of the hour) on some channels.

A raw file is at 
https://we.tl/t-I32wFGP5H1
(thanks to user DanDAB) 

These services play without problems in welle.io

BBC Radio 1 (0xc221) Pri subch= 1 start= 0 CUs= 96 PL=uep 3 bitrate=128
BBC Radio 2 (0xc222) Pri subch= 2 start= 96 CUs= 96 PL=uep 3 bitrate=128
BBC Radio 3 (0xc223) Pri subch= 3 start=192 CUs=140 PL=uep 3 bitrate=192
(it's playing also with 160 kBit compared to Andi's DAB player)

There is no sound after re-config for

BBC Radio 4 (0xc224) Pri subch= 4 start=332 CUs= 96 PL=uep 3 bitrate=128
BBC Radio 5 Live (0xc225) Pri subch= 5 start=428 CUs= 58 PL=uep 3 bitrate=80
BBC Radio 1Xtra (0xc22a) Pri subch=10 start=582 CUs= 96 PL=uep 3 bitrate=128
BBC AsianNetwork (0xc236) Pri subch= 7 start=486 CUs= 48 PL=uep 3 bitrate=64
BBC WorldService (0xc238) Pri subch= 9 start=534 CUs= 48 PL=uep 3 bitrate=64
BBC Radio 4Extra (0xc22c) Pri subch=12 start=774 CUs= 58 PL=uep 3 bitrate=80
BBC Radio 6Music (0xc22b) Pri subch=11 start=678 CUs= 96 PL=uep 3 bitrate=128

See logfile attached
[welleio.log](https://github.com/AlbrechtL/welle.io/files/4996709/welleio.log)

AlbrechtL, 2020-09-03:
Thanks for reporting. Please upload your file again.

andimik, 2020-11-24:
Sorry for the delay, did not read your answer.

It's now on my homepage, so will not be deleted after 7 days.

http://andimik.bplaced.net/temp/12B_BBC%20National%20DAB_2020-07-27_095947.zip

AlbrechtL, 2020-12-08:
Thanks for the file. Now I can reproduce the issue.

Do you know when the BBC will switch to DAB+?

nickpiggott, 2021-02-24:
The reconfiguration process is entirely normal and part of the standard. In the run-up to reconfiguration, the new configuration is sent interleaved with the current configuration (for at least 15 frames prior to the change), and there's a frame counter to tell you exactly which frame to change over on. I wouldn't call it Frankenstein.

The BBC has never made any comment on a timeline for changing to DAB+. It's possible they'll be the last to change and only change some of their services.

AlbrechtL, 2021-05-22:
@nickpiggott Thanks for your explanation. Because I'm living in Germany I'm not effected and thus this bug doesn't have high priority on my list.

But I will leave this issue open because we shall fix it.

Pull requests are welcome!

AlbrechtL, 2024-10-08:
Does the BBC still broadcasts via DAB?

AlbrechtL, 2024-10-09:
> The reconfiguration process is entirely normal and part of the standard. In the run-up to reconfiguration, the new configuration is sent interleaved with the current configuration (for at least 15 [...]

@nickpiggott Can you point me to that direction where to find this information within the frame?

nickpiggott, 2024-10-09:
Hello

It's in section 6.5 (Multiplex Reconfiguration) of [EN 300 401](https://www.etsi.org/deliver/etsi_en/300400_300499/300401/02.01.01_60/en_300401v020101p.pdf).

The C/N flag on the relevant FIGs indicates if that FIG refers to the Current or Next configuration of the multiplex. For example, the ordering of subchannels and the start of each channel may change at a reconfiguration.

AlbrechtL, 2024-10-09:
Thanks for that. I just saw that the backend already detects the reconfiguration but is not handling it :-(.
https://github.com/AlbrechtL/welle.io/blob/330ee0a16318ccb9cb2f70d69cf325feaa143ccb/src/backend/fib-processor.cpp#L144-L157

Maybe an option is simply retuning to the service. This is not nice but should work.

AlbrechtL, 2024-10-09:
OK, I added a workaround in https://github.com/AlbrechtL/welle.io/commit/c0434be71fe45c1ca8da9946dc44b5b828e56c84. 

It is not a nice solution because it just restarts the decoder, which creates a short audio drop. But it is better than nothing :-).

Please test https://welle-io-nightlies.albrechtloh.de/20241009_c995c85b_Windows_welle-io-setup_x64.exe.

beermad, 2024-10-11:
Thanks for this attempt. Unfortunately it seems not to work.

Log shows output from a test run, with audacious streaming the audio on one channel.
I hope this gives some useful clues; if there's any more information you need, just ask - is there a debug mode that would give _more_ useful data?

[welleTest.log](https://github.com/user-attachments/files/17345132/welleTest.log)

AlbrechtL, 2024-10-11:
Thanks for the test. I reopened the issue.

Since I'm living in Germany, I can't try this by my own. I did the test with the recording by @andimik.

Can you provide me a RAW file recording? In the expert mode you can do a RAW file recording.

beermad, 2024-10-11:
> Can you provide me a RAW file recording? In the expert mode you can do a RAW file recording.

Will do. Is that from the option "Enable Tll decoding to console log"? Or is some other option I haven't noticed needed?

andimik, 2024-10-12:
No, please have a look at https://www.welle.io/devices/rawfile

AlbrechtL, 2024-10-12:
You can also try the experimental RAW file recorder
[image]

But you have to catch the reconfiguration. So the best would be to have a sample with +-5 minutes around the reconfiguration.

beermad, 2024-10-12:
> No, please have a look at https://www.welle.io/devices/rawfile

Great, thanks.
The next reconfiguration I can confidently predict will be late on Monday afternoon. I'll record that and upload it.

beermad, 2024-10-14:
Raw DAB dump (uploaded to cloud as it's a bit big to do in browser) [https://filedn.com/lnJp6X7QX8y063Wuq1QkInV/20241014_165850_12B.iq](https://filedn.com/lnJp6X7QX8y063Wuq1QkInV/20241014_165850_12B.iq)

Output from `welle-cli -f 20241014_165850_12B.iq -p "BBC Radio4"` (using the welle-cli built from your updated files). 
[log.txt](https://github.com/user-attachments/files/17367391/log.txt)

AlbrechtL, 2024-10-14:
Thanks, I got the file!

andimik, 2024-10-14:
but the SNR is very bad, hard to get errorfree audio.

beermad, 2024-10-14:
> but the SNR is very bad, hard to get errorfree audio.

I was surprised how poor the sound quality was after recording the stream using your dab_raw_record.sh script. Though something seemed a bit odd about it, as it ignored my -t argument and I had to kill it manually after the sync failure had occurred.

I wonder if the hard-coded sample rate might need changing, though when I tried doubling the rate it didn't work.

I may try again, dumping from welle-io when the problem recurs tomorrow evening if necessary.

andimik, 2024-10-14:
You could also use 

- https://github.com/JvanKatwijk/qt-dab/releases/  (save as xml file, this is smaller than sdr)
- https://github.com/KejPi/AbracaDABra/releases (see Ensemble Information menu)

beermad, 2024-10-15:
Another attempt at a dump, this time using welle-io.
[https://filedn.com/lnJp6X7QX8y063Wuq1QkInV/welle-io-record.iq](https://filedn.com/lnJp6X7QX8y063Wuq1QkInV/welle-io-record.iq)

I think it stopped recording pretty much as soon as the sync was lost, so just have to hope it managed to get what you need. Much better quality this time round.

andimik, 2024-10-16:
Yes, much much better (21 dB)!

AlbrechtL, 2024-10-30:
@beermad Thanks for the new file. I just checked it. The SNR is fine now, but I don't see a reconfiguration included.

beermad, 2024-10-30:
> @beermad Thanks for the new file. I just checked it. The SNR is fine now, but I don't see a reconfiguration included.

Hmmm... Sorry about that. I was pretty sure I'd caught it. I'll have another try and actually _check_ it before uploading it.

beermad, 2024-10-31:
This may take a bit longer than I'd hoped. The regular 1800 reconfiguration I was relying on hasn't happened for a few days. I'm having to do some work to find out another time I can rely on.

beermad, 2024-11-06:
Apologies for the delay; it seems the BBC's schedule for reconfiguring its multiplex has changed, so I could no longer rely on what was the regular 1700 reconfiguration.
New recording uploaded to my loud server](https://filedn.com/lnJp6X7QX8y063Wuq1QkInV/BBC.uff).

I've double-checked and there's definitely a reconfiguration here. AbracaDABra (which I recorded it with) took the change in its stride, but playing back the recording with well-cli I get the usual crash.

andimik, 2024-11-06:
Yes, BBC Radio 5 gets a new subchannel.

```
0xC225 BBC Radio5Live [ BBC R5L ] ECC: 0xE1, Country: United Kingdom, PTy: News (static), Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'BBC Radio5Live' [ 'BBC R5L' ], ASCTy: 0x0 (MP2)
      SubChId: 5, Language: English, StartCU: 428, NumCU: 58, UEP #21, Protection level: 3, Bitrate: 80kbps
```

becomes

```
0xC225 BBC Radio5Live [ BBC R5L ] ECC: 0xE1, Country: United Kingdom, PTy: News (static), Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'BBC Radio5Live' [ 'BBC R5L' ], ASCTy: 0x0 (MP2)
      SubChId: 5, Language: English, StartCU: 404, NumCU: 48, UEP #16, Protection level: 3, Bitrate: 64kbps
AudioComponent (secondary), SCIdS: 3, Label: 'BBC Radio5SX' [ 'BBC R5SX' ], ASCTy: 0x0 (MP2)
      SubChId: 8, Language: English, StartCU: 500, NumCU: 48, UEP #16, Protection level: 3, Bitrate: 64kbps
```

beermad, 2024-11-27:
For info, I've been monitoring this more closely than usual (mainly so I can work out the new pattern of reconfigurations to schedule any future tests). I've found three distinct error messages appearing:
`Big change (MPEG version, layer, rate). Frankenstein stream?` (As given in the original bug report)
`Illegal Audio-MPEG-Header 0xf0d044c4 at offset nnnnnn`
`Hit end of available data`

Cheers.

AlbrechtL, 2024-11-27:
These error messages are coming direct from the `mpg123` library and are indicating that the stream is broken. I mean that's correct because the stream is broken. What we have to do is to restart the decoder once the reconfiguration is detected. In https://github.com/AlbrechtL/welle.io/commit/c0434be71fe45c1ca8da9946dc44b5b828e56c84 I tried to restart the decoder but it seems that this is not working :-(.

It is possible for you to provide a IQ RAW file with included reconfiguration?

AlbrechtL, 2026-10-11:
Thanks to AI reconfiguration should be supported now (see https://github.com/AlbrechtL/welle.io/commit/b5e669167edca51f8c88eb2f1d284e166caf8400).

beermad, 2026-10-14:
> 30\. Thanks to AI reconfiguration should be supported now (see [b5e6691](https://github.com/AlbrechtL/welle.io/commit/b5e669167edca51f8c88eb2f1d284e166caf8400)).

Thank you!

andimik, 2026-10-14:
Can't receive BBC, but let's close it.

I could test it with RAI Piedmont West via Satellite feed, but it's just a few times a day when the reconfiguration happens.

closed 2026-10-14 by andimik