AlbrechtL/welle.io #589: How to connect hackrf one

langxj, 2020-08-08:
First of all, thank you for writing some outstanding software! I have a hackrf one board that I can operate with SDR#. How to use it welle.io Operation hackrf one

gvanem, 2020-08-08:
HackRF is supported by the [SoapySDR API](https://github.com/pothosware/SoapySDR/wiki). 
Welle-IO can be built with support for SoapySDR. It depends on which Welle-IO you have and how it was built. 

Then select SoapySDR as the source in the settings-box:
[image]

You could build it yourself to ensure it uses the SoapySDR API.

AlbrechtL, 2020-09-03:
@langxj Which OS you are using?

closed 2024-10-08 by AlbrechtL