AlbrechtL/welle.io #594: "No valid device found use Null device instead" on Raspberry pi 4 Buster

bowosa, 2020-09-26:
I am trying to get my USB "Nooelec NESDR Nano 2" to run with welle.io. I am running the latest OpenAuto Pro 10 distribution which has the welle.io 2.0-Beta 2 installed. I have blacklisted the "dvb_usb_rtl28xxu" but welle.io will not detect the device.
Executing rtl_test confirms the device is well detected and runnning: 
**_pi@raspberrypi:~ $ rtl_test
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6
[R82XX] PLL not locked!
Sampling at 2048000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...
Allocating 15 zero-copy buffers
lost at least 32 bytes_**

But as soon as I run welle.io-2 no device can be found!Does anybody know what might be going wrong? 
Any advice would be greatly appreciated!

AlbrechtL, 2020-09-30:
Can you post the console output of welle.io please?

bowosa, 2020-10-01:
Hello Albrecht,Greatly appreciate your reply. Unfortunately I have returned the device as I still had the chance.  Before buying a new one can you recommend one! In the meantime would a console output still be helpful? If yes could you tell me how you generate the console output? Thanks for your support.

bowosa, 2020-10-10:
Hi Albrecht, 
got a new USB rtl-sdr devcie. Could you again give details how I shall provide the console output of welle.io to you?

bowosa, 2020-10-12:
When launching welle-io-2 I get follwoing logs: 
welle-io-2
2020-10-12T16:28:04.057 Debug: using qt5ct plugin
2020-10-12T16:28:04.231 Debug: D-Bus system tray: no
2020-10-12T16:28:04.233 Debug: Audio: ActiveState
2020-10-12T16:28:04.558 Debug: main: Set language "de_DE"
2020-10-12T16:28:04.576 Debug: Apply settings initially
2020-10-12T16:28:04.576 Debug: RadioController: Close device
2020-10-12T16:28:04.578 Info: InputFactory:Input device:auto
2020-10-12T16:28:04.648 Debug: RadioController: AGC on
2020-10-12T16:28:04.649 Debug: Audio: Volume 1
2020-10-12T16:28:04.649 Warning: qrc:/QML/settingpages/GlobalSettings.qml:176: TypeError: Property 'initDevice' of object QQuickItem(0x2187590) is not a function
2020-10-12T16:28:04.788 Warning: file:///usr/lib/arm-linux-gnueabihf/qt5/qml/QtQuick/Controls.2/ApplicationWindow.qml: Conflicting properties 'visible' and 'visibility' for Window 'window'
2020-10-12T16:28:04.875 Debug: os: linux
2020-10-12T16:28:04.875 Debug: desktopAvailableWidth: 800
2020-10-12T16:28:04.876 Debug: desktopAvailableHeight: 428
2020-10-12T16:28:04.876 Debug: orientation: 2
2020-10-12T16:28:04.876 Debug: devicePixelRatio: 1
2020-10-12T16:28:04.876 Debug: pixelDensity: 3.7854983766839574
2020-10-12T16:28:04.877 Warning: qrc:/QML/GeneralView.qml:60: TypeError: Cannot call method 'createObject' of null

bowosa, 2020-11-16:
Dear Albrecht,Please excuse for again asking for your help. I am still failing to get my rtl-sdr usb dongle to be recognized by my raspberry pi Welle.io installation.  I you could provide me some support to fix my issue I would greatly appreciate it. With best regards,BorisBoris <bowosa@gmx.net> schrieb am Do. 1. Okt. 2020 um 11:24:Hello Albrecht,Greatly appreciate your reply. Unfortunately I have returned the device as I still had the chance.  Before buying a new one can you recommend one! In the meantime would a console output still be helpful? If yes could you tell me how you generate the console output? Thanks for your support.

AlbrechtL, 2020-11-21:
Your console output is too short. Please the complete output.

bowosa, 2020-11-22:
Hi Albrecht, the output I pasted was all what was available. Can you tell which command I need to enter to get a full output?

bowosa, 2020-11-30:
I progressed with the Welle.io installation. I guess I did a mistake by not correctly finalizing the installation. Now welle.io starts and detects my Realtek TRL2838UHIDIR USB dongle. The problem I now have is that when launching a scanning no channels are found. During the scan process the Frame sync switches often between "OK" and "Not synced" and the red antenna icon turns from red to black but no channels are detected.
Any idea what I could be the issue here?

AlbrechtL, 2020-12-02:
It looks that your reception is too bad.

bowosa, 2020-12-04:
I just tested the device with Windows10 using CubicSDR, Here I am definitely getting some Radio signals
[image]
I noticed that this program seems to use my device via a SoapySDR options. Maybe on Welle.io I should use this as well. I will give it a try!

bowosa, 2020-12-05:
SoapySDR was not successfuill but when I switched abck to rtl_sdr setting and increased the gain to around 20 then welle.io started to detect channels. Hurrrayy! However, when selecting a channel there is no sound. I have a USB soundcard attached which works well when I connect via my Mobile Phone via bluetooth to stream spotify.

andimik, 2020-12-05:
This screenshot does not help as it shows FM band only.

Please repeat it showing the DAB channel found in Welle.io.

andimik, 2020-12-05:
If not, use another (external) antenna.

bowosa, 2020-12-06:
Hi andimik,

Thanks for the advice. I seem now however to have solved the issue. I figured after a while that the "gain" settings was much to low. When I started to increase the gain up to around 15 the scan started to detect channels. Really too bad that this issue nobody seem to have addressed before as I could have saved quite some time de-bugging this issue. Nevertheless, thanks for your support!

andimik, 2020-12-06:
No, you did not notice Albrecht's hint four days ago.

AlbrechtL, 2021-03-06:
> I just tested the device with Windows10 using CubicSDR, Here I am definitely getting some Radio signals

Please tune to your DAB+ frequency not the FM frequencies.

AlbrechtL, 2024-10-08:
I'm closing this issue because I'm not sure if it is still relevant. If it still occurs in the latest version, we can reopen it.

closed 2024-10-08 by AlbrechtL