AlbrechtL/welle.io #595: folder depth for MOT image saving

Observerist, 2020-10-01:
Great to to be able now do target a folder.

But besides the default Home folder as used before, I can only go one level further, e.g. my Desktop can be used, but not one level deeper where a specified folder is not being validated. 

Am I correct in believing that images cannot be saved continuously as they are, one after the other, pushed on air ?
While only pictures that are already in the cache can be retrieved, as a batch if necessary ?

andimik, 2020-10-07:
It's depending if the service you are listening to uses different file names or not.

AlbrechtL, 2020-12-02:
> But besides the default Home folder as used before, I can only go one level further, e.g. my Desktop can be used, but not one level deeper where a specified folder is not being validated.

Which OS you are running?

>Am I correct in believing that images cannot be saved continuously as they are, one after the other, pushed on air ?

Correct

>  While only pictures that are already in the cache can be retrieved, as a batch if necessary ?

Yes

AlbrechtL, 2024-10-08:
Closed because of inactivity.

closed 2024-10-08 by AlbrechtL