AlbrechtL/welle.io #596: Audio drops while driving

m1k6, 2020-10-02:
Hello everyone,
I own an android head unit (radio) for my car and tried out welle.io.
In itself it works, but quite often the audio drops. SNR is very good, most of the time I have 3-4/4 green bars.

I also noticed, after audio dropped, that the frequency corrector retries to "find" the exact frequency again. I use an RTL2832U from RTL-SDR.com which has very low ppm, so I disabled the coarse corrector.
Most interestingly, I tried out "Wavesink" (an old android app you have to pay for) and on the same route, same channel (5C here in Germany), I did not have a single drop of audio.
There were no other apps running, both were running in foreground and I tested this on multiple days, always with welle.io having audio dropping and Wavesink working perfect.

The head unit has a quad core ARM 1,5 Ghz, so I _guess_ CPU power isn't an issue.

AlbrechtL, 2020-12-02:
Thanks for your report.

Mostly likely our software auto gain control (AGC) of the rtl-sdr is too slow. Also others reported to me that wavesink performances is better compared to welle.io. We definitely have space for improvements.

AlbrechtL, 2024-10-08:
Closed because of inactivity.

closed 2024-10-08 by AlbrechtL