AlbrechtL/welle.io #597: Get the Car Information/ EPG out of the Stream?

nousername1, 2020-10-07:
Hi
Attlest in Austria 2x interesting stuff get transmit with the Audio Channel:
https://www.rtr.at/de/m/MUXI

Electronic Program Guide | ORS comm GmbH & Co KG | ZD | 12 |   |   |   |   | **0xE0A0AD50**
Verkehrsinformationen (TPEG) | ORS comm GmbH & Co KG | ZD | 6 |   |   |   |   | **0xE0A0AD51**

on the other MUX:

Electronic Program Guide (EPG) | RTG Radio Technikum GmbH | ZD | 12 |   |   |   | X | **0xE0A0AD20**
Emergency Warning Function (EWF) | RTG Radio Technikum GmbH | ZD | 24(**) |   |   |   | X | **0xE0A0AD21**

Thanks

andimik, 2020-10-07:
This page only shows what is allowed. TPEG is not used in Austria and not supported by Welle.io.

A complete overview is on http://www.fmlist.org/tpeg.php

AlbrechtL, 2020-12-02:
About SPI see here, please: https://github.com/AlbrechtL/welle.io/issues/558
About TPEG see here, please https://github.com/AlbrechtL/welle.io/issues/501

In short: Both are no possible at the moment.

closed 2020-12-02 by AlbrechtL