AlbrechtL/welle.io #600: Windows GUI - well-io.exe - Howto use rtl_tcp - unknopwn options

Olivier6767, 2020-10-17:
Hello,
I'm trying to use well-io.exe (GUI) on windows to access my RTL_SDR device which is running on a raspberry and shared using rtl_tcp.
I've found this example in an older forum thread: `welle-io.exe -d rtl_tcp --rtltcp-ip 10.1.1.91 --rtltcp-port 8100`
However, regardless of what command line options I try, I always get: `welle.io: Unknown options: d, rtltcp-ip, rtltcp-port`
Also, `welle-io --help` does not dhow any option related to device selection
Am'I missing something? 
What is the correct way of using rtl_tcp with welle-io ?

Thanks!

Akito01, 2020-10-30:
Go for version 1.0 . I had the same issue. In 1.0 it works: -d rtl_tcp --rtl_tcp-address 10.1.1.91

Olivier6767, 2020-10-31:
Thanks. I tried version 1.0 and it does indeed connect to `rtl_tcp`.
However I cannot find any stations, tried setting manual Gain and AGC... 
I can see in the `rtl_tcp` log that it does change the frequencies, so the communication between `welle.io` and the `rtl_tcp` instance is working.
When using SDR# with the same `rtl_tcp` connection, I can access FM stations fine.
Any idea what else I can try ?

Is there a plan to support `rtl_tcp` on version 2+ ?

closed 2020-10-31 by Olivier6767