AlbrechtL/welle.io #602: DLS infoes on MP3 stream

FdeAlexa, 2020-11-02:
Hello to all,
I normally use welle.cli as a server streamer on Raspberry pi.
I decode the streaming on a PC using VLC and, thanks to the issue #559 ([welle-cli] Allow switching channel on get request), I can listen to the DAB + radio through the typical request "http://MYHOSTIP:7979/mp3/CHANNEL/SID".
I also decode the stream with the "moOde AUDIO" interface.
The REQUEST is:
Is it possible to receive DLS information (title, author, type, etc.) such as RDS in FM or METADATA data in Internet RADIO?
I am not sure if there is already an option to receive this information.
Would you please be so kind to help me on this issue?
Best regards and thanks,
Francesco

spongioblast, 2021-07-20:
Would love to see this feature or to know if there is any way. Thanks!

mpbraendli, 2024-10-09:
Closing due to lack of activity.