AlbrechtL/welle.io #603: SoapySDR remote server not supported?

Akito01, 2020-11-05:
I've been trying to get welle.io 2.2 for Windows to work with SoapySDR but I came to realize that there's no remote setting.
I'm running SoapySDRServer on a remote computer with an Airspy. Will it be possible to use this in Windows soon remotely?

gvanem, 2020-11-05:
Have you tried connecting to it using `SoapySDRUtil.exe --make=driver=remote ...`?  Not sure about the address syntax for this.
But if this works, it should work in Welle-IO too.

Akito01, 2020-11-05:
I hadn't tried that, but now I have... though Welle-IO throws: Error while opening device.
I have SoapySDRServer running like: `SoapySDRServer --bind="0.0.0.0:1234"` on my remote Linux machine.

On my Windows machine running: `soapysdrutil --find="remove=tcp://192.168.2.2:1234"`
I see this:
```
Found device 4
  driver = remote
  label = AirSpy One [myserial]
  remote = tcp://192.168.2.2:1234
  remote:driver = airspy
  serial = myserial
```

And SoapySDRServer shows an accept and close for the connection.
So there is connectivity.
In Welle-IO I have chosen SoapySDR and a driver setting like: `driver=remote,remote=tcp://192.168.2.2:1234`

AlbrechtL, 2020-12-02:
Honestly, I never tested SoapySDR remote and I never tested SoapySDR under Windows at all. Generally, it should work because SoapySDR tries to be generic. But I know from the Soapy support from other hardware that peoples submitted patches to get it work.

astro, 2025-02-10:
I really wanted to try this but our SDR is already installed on the roof so I put the parameters verbatim into the source code. Perhaps that will help anyone in a similar situation:

```patch
diff --git src/input/soapy_sdr.cpp src/input/soapy_sdr.cpp
index cef8c787..366a1524 100644
--- src/input/soapy_sdr.cpp
+++ src/input/soapy_sdr.cpp
@@ -49,7 +49,7 @@ CSoapySdr::CSoapySdr(RadioControllerInterface& radioController) :
 {
     //enumerate devices
     const std::string args ="";
-    const auto foundDevices = SoapySDR::Device::enumerate("");
+    const auto foundDevices = SoapySDR::Device::enumerate("remote:driver=hackrf,remote=tcp://172.20.73.57:55132");

     int nonAudioDeviceCount = 0;
     for(auto found: foundDevices)
@@ -99,7 +99,7 @@ bool CSoapySdr::restart()
     m_spectrumSampleBuffer.FlushRingBuffer();

     try {
-        m_device = SoapySDR::Device::make(m_driver_args);
+        m_device = SoapySDR::Device::make("driver=remote,remote:driver=hackrf,remote=tcp://172.20.73.57:55132");
     }
     catch (std::exception& e) {
         std::clog << "Exception caught in SoapySDR::Device::make with \"" << m_driver_args << "\" :\n" <<
```

AlbrechtL, 2025-03-07:
Thanks for posting a workaround.

Did you try to use the field "Driver arguments"?
[image]

AlbrechtL, 2025-03-07:
OK, you found it. But just for reference here is the explanation image

[image]