AlbrechtL/welle.io #605: wrong Parameter for rtl_tcp connection at Version >V2.0

messschuh, 2020-11-20:
Tried to connect to rtl_tcp server with Parameters from welle.io Webpage (welle-io.exe -d rtl_tcp --rtl_tcp-address 10.10.1.10 --rtl_tcp-port 1235) and got response  "welle.io:Unkown option D, rtl_tcp-adress, rtl_tcp-port) .
I tried V2.2 and V2.1.
also tried V1.0 with Parameters -d -i -p , that was working fine. it seems parameters has changed, so what are the correct Parameters from Version 2.0?

AlbrechtL, 2020-12-09:
Thanks for the hint. Unfortunately, the web page was outdated. I fixed it in https://github.com/AlbrechtL/welle.io/commit/8a5b0e0549d344966d7c57d683d54dac4f3fc2df.

Please check https://www.welle.io/devices/rtl_tcp again.

closed 2020-12-09 by AlbrechtL

andimik, 2020-12-10:
Is there any special reason why it's called rtl-tcp instead of (commonly used) rtl_tcp?

AlbrechtL, 2020-12-10:
No there is no real reason.