AlbrechtL/welle.io #608: Is CIR properly tool-tipped?

pswiatki, 2020-12-02:
I added something in dc888e2f7daa6c7ea512de818cb7e2fe5198f166.

Do you mean this?
[image]

_Originally posted by @AlbrechtL in https://github.com/AlbrechtL/welle.io/issues/309#issuecomment-435099798_

pswiatki, 2020-12-02:
Also: the following is mentioned:

> Conversion is:
> 
> * `microseconds` = `samples` / 2.048
> * `km` = `microseconds` × 3e5

pswiatki, 2020-12-02:
By the way: why does the peak in CIR (start of symbol, I presume) always fall on (or near?) sample 512?
Moreover: shouldn't the delay and distance be offset off that peak instead of simply being based on a sample number? That would allow for an easier estimation of the delay/distance e.g. in SFN reception scenarios, or just when heavy multipath is experienced.

pswiatki, 2020-12-02:
Oh, one more: zoom on the X-axis would be perfect - when delays are short it is not easy to actually see the individual peaks.

AlbrechtL, 2020-12-02:
@mpbraendli Can you answer pswiatki questions?

mpbraendli, 2020-12-04:
Yes it's indeed a speed-of-light conversion. I agree fixed instead of scientific notation would look better.
The fact that the peak is always around 512 is a consequence of some internals. It would be more logical to present this at 0 (and calculate the timings from that). You're anyway not doing absolute measurements, only one peak relative to another.

Do you want to make a pull request with these changes? I encourage you to try!

AlbrechtL, 2021-03-06:
Pull request are welcome. In the mean time I'm closing this issue.

closed 2021-03-06 by AlbrechtL