AlbrechtL/welle.io #612: US Digital Radio Support

andrewfer000, 2020-12-20:
I saw this issue before but I wanted to bring it up again. This is great for when want to listen to international DAB broadcasts using other people's SDRs over rtl-tcp but I would also like to use my own SDR at home to listen to "HD Radio" aka NRSC-5 based broadcast. Currently there is [this tool](https://github.com/theori-io/nrsc5) that is cross platform and works well for HD Radio signals with my SDR but the problem is that it's command line only and not touch screen friendly and I would like to see the information such as Title, Artist, Description, as well as Album Art (if possible) just like on DAB. NRSC-5 also is capable of getting local weather, traffic, and emergency alert information which would make this program super awesome for a road warrior like myself!

Extra Infos:
 [NRSC-5 standard](https://www.nrscstandards.org/standards-and-guidelines/documents/standards/nrsc-5-d/nrsc-5-d.asp)
 [HD Radio for US and Canada ](https://hdradio.com/)

andimik, 2020-12-20:
The main problem is that currently all programmers and users of Welle.io have no HD radio (which is only used in a very few countries).

Therefore, I fear you have to find someone in your country who forks the project and implement your wishes.

andimik, 2020-12-20:
> Welle.io is great for when want to listen to international DAB broadcasts using other people's SDRs over rtl-tcp

No, this is not the intention (it works, but generate lots of traffic).

Welle.io listeners are using their ****own****  SDR of course.

andrewfer000, 2020-12-20:
Yea I guess you can't really implment a fearure you don't have access to xD.

 Also, my rtl-tcp situation is a little different. My British friend and I have gigibit internet and low latency fiber connections so using Welle.io over the internet isn't as bad as you may think.

andrewfer000, 2020-12-20:
Also, If someone is able to implment this in the future, then I'm willing to test it.

marcusmueller, 2021-03-03:
@andrewfer000 I think the fair thing to point out here is that this would be really really really significant part of the code base of welle.io that needs to be adapted; NRSC-5 and DAB+ aren't *that* similar in a lot of technical aspects. 

As someone who's maintained projects before: your best guess in a "large feature request" situation like this is to *become the expert you want the world to see* (a.k.a, a hero). It's not that hard: You can use 
https://github.com/argilo/gr-nrsc5 to generate a NRSC-5 signal, and then simply pipe it via named pipe to its IQ file input or via TCP, emulating `rtl_tcp` into your https://github.com/theori-io/nrsc5/ ; that software is also GPL, and you can gradually move its functionality into welle.io. 

The actual hardship stems from putting everything behind the same semantics, make it transparently useable and selectable what you want to receive, and be the person to contact w.r.t. testing things on real NRSC-5 transmissions.

All in all, I'd very respectfully suggest to consider this issue as "out of scope until someone contributes the code". But I'm not a welle.io maintainer, and it's up to one of those to categorize things :)

AlbrechtL, 2021-03-06:
@andrewfer000  Thanks for your request!

As discussed in https://github.com/AlbrechtL/welle.io/issues/340 we need developers for doing this.

closed 2021-03-06 by AlbrechtL

andrewfer000, 2021-03-31:
If anyone is looking for a GUI for HD Radio in the US-
https://github.com/markjfine/nrsc5-dui