AlbrechtL/welle.io #614: [welle-cli] DAB channel DLS info not updated

glokhoff, 2020-12-22:
I am noticing that the DLS information on the webpage generated by welle-cli is updated for every change of the DLS text on the DAB+ channels, but not on the single DAB channel we have in our local multiplex. (We're still offering this channel as in the past quite a large number of DAB receivers were sold in our region, and hope to increase the number of listeners this way.)

andimik, 2020-12-22:
So do you fear it's related to DAB or to the single service?

glokhoff, 2020-12-22:
I'm assuming DAB, but could be wrong... If I tune to another channel and retune to our local multiplex the info gets updated.

andimik, 2020-12-22:
I've tried with BBC and can confirm it works with DAB.

glokhoff, 2020-12-22:
Yes, until yesterday I had changing DLS text for each of the stations in the DAB channel, and did not notice a problem. However, I changed this yesterday to a single DLS text line for each of the 2 stations in the DAB channel, and noticed the displayed DLS information did not change at the moment of switch over from one to another. I assume somehow welle-cli only checks for new text on discovering the change (perhpas through some flag in the DAB side info ?). Regular receivers seem to continuously check for new DLS info, as they do show the switch over.

andimik, 2020-12-22:
Could you share an eti file?

glokhoff, 2020-12-22:
Ouch... I have to see how to generate/capture that... Also at the moment of switch over, I assume ? Is there a welle.io mode that captures this into an eti file ?

andimik, 2020-12-22:
Please use https://github.com/JvanKatwijk/eti-stuff to generate it.

But of course a raw file is good as well (but much larger).

glokhoff, 2020-12-22:
OK, that requires a bit of work on installing eti-cmdline. And as I'm not a C programmer, and have little experience compiling programs, this will have to wait until after Christmas.... Happy holidays for now !

AlbrechtL, 2021-03-27:
Any progress here?

glokhoff, 2021-03-28:
Sorry, no - in the meantime the DAB channel has been dropped from the multiplex in favor of a new DAB+ station...

andimik, 2021-03-28:
Where are you located?

glokhoff, 2021-03-28:
Eindhoven, The Netherlands

andimik, 2021-03-30:
Maybe you ask Radio4Brainport itself.

The problem is, that on the welle-cli monitor on http://185.238.128.xxx:7979/ they are using a too old version.

I recommend the next branch, where SLS and other things have been fixed already.

glokhoff, 2021-04-02:
Well, I'm running that site myself ;O) 
When I try to update the version with sudo apt install welle.io it says it is also the latest version (2.1.-2build1) ...
It's on a Linux Mint 20 machine, so I went with the instructions for Ubuntu.

andimik, 2021-04-02:
I just tried with the appimage and it works.

Well, you've mentioned another (still unsolved) issue. Please change to the directory where the `index.html`, the `index.js` and the welle-cli app is located or copy both files.

If you cannot find it on your harddisk, please download both from https://github.com/AlbrechtL/welle.io/tree/next/src/welle-cli

BTW: Please use rtltcp for better performance.

# in one console 
start `rtl_tcp`

# and in another
`./20210306_6e4026b_Linux_welle-io-cli-x86_64.AppImage -Dw 7979 -F rtl_tcp,127.0.0.1:1234`
(whereas in this folder `index.js` and `index.html` are located as well)

Should look like

[image]

andimik, 2021-04-20:
Good morning, I just recognized you've upgraded the welle-cli installation.

AlbrechtL, 2021-05-22:
> ./20210306_6e4026b_Linux_welle-io-cli-x86_64.AppImage -Dw 7979 -F rtl_tcp,127.0.0.1:1234

Cool, for me this is the first note that somebody is using the welle-cli AppImage ;-)

I think this issue is solved and I'm closing it.

closed 2021-05-22 by AlbrechtL