AlbrechtL/welle.io #615: [welle-cli]

d2009, 2020-12-22:
Hi there,
is it possible to operate several RTLSDR USB sticks with -cli?
And: would it be possible to install a DAB scan in -cli?

andimik, 2020-12-23:
Yes, this is possible, I just tried it.

# First dongle

`$ rtl_tcp -d 0`
`$ ./welle-cli -F rtl_tcp,127.0.0.1:1234 -Dw 7980`
http://localhost:7980/

# Second dongle

`$ rtl_tcp -d 1 -p 1250`
`$ ./welle-cli -F rtl_tcp,localhost:1250 -Dw 7979`
http://localhost:7979/

Please note, you need a powerful PC if you are using parameter `D`

And regarding the scan function: You could change the file `index.js` (in your build-directory) by deleting the channels without any signal in your region:

```
var channels = [ "5A", "5B", "5C", "5D", "6A", "6B", "6C", "6D", "7A", "7B",
"7C", "7D", "8A", "8B", "8C", "8D", "9A", "9B", "9C", "9D", "10A", "10B",
"10C", "10D", "11A", "11B", "11C", "11D", "12A", "12B", "12C", "12D", "13A",
"13B", "13C", "13D", "13E", "13F"];
```

becomes for example

```
var channels = [ "5C", "12B", "12C", "12D"];
```

Then the list is much shorter.

AlbrechtL, 2021-03-27:
@andimik Thanks for the workaround!

Duplicated with https://github.com/AlbrechtL/welle.io/issues/95.

closed 2021-03-27 by AlbrechtL