AlbrechtL/welle.io #616: headless server

baloe, 2020-12-25:
Hey I have a GNU/Linux-based headless server with MPD running for playing music. I remote-control it with an MPC app on my Android device.

Now, I'd like to extend its capabilities to DAB+ by installing a software like welle or dabd, but I was unable to find any working solution that also provides a means of remote-controlling it, thinking of a web-interface or a remote-control app.

I am currently considering writing a web-application that would run on the machine to control dabd, which could then be accessed from any device, but I'd be happy to stick with a simpler solution. I'd be happy to hear your advice.

andimik, 2020-12-25:
What is dabd? I find too much results on github.

baloe, 2020-12-26:
Ah I'm sorry, I mixed something up; not dabd, but https://github.com/schlizbaeda/DABradio

mpbraendli, 2020-12-27:
Have a look at welle-cli and its integrated web interface. See https://github.com/AlbrechtL/welle.io#welle-cli

beermad, 2021-02-12:
> Hey I have a GNU/Linux-based headless server with MPD running for playing music. I remote-control it with an MPC app on my Android device.
>
> Now, I'd like to extend its capabilities to DAB+ by [...]

welle-cli is definitely the route to go down. I have a custom-built radio timeshifting system which relies completely on it to decode the DAB multiplexes.

By running it with the web interface enabled (using -w PORTNUMBER) you can either connect to it with a browser (http://hostname:PORTNUMBER) and can play any channel from the mux it's tuned to by pointing any player (or recorder, in my case) at http://hostname:PORTNUMBER/mp3/SID
I recommend only running welle-cli when you actually want to use it, as the nature of SDR is that it's quite CPU-intensive.

closed 2021-03-27 by AlbrechtL