AlbrechtL/welle.io #617: Driver Arguments and Antenna for B210?

PE1NUT, 2021-01-01:
What 'driver arguments' and 'antenna' values should one use with the B210 USRP? Despite a lot of searching, I've been unable to find any information on this.

Welle.io does seem to work with my B210 (together with SoapySDR and its UHD plugin), as the spectrum window shows the received signal. But given the huge DC spike in the centre, it doesn't seem to be configuring the antenna port of the B210 correctly. This is also clear because none of the front panel lights of the B210 are on.

OS is Ubuntu 20.04, with Welle.io 2.1 and SoapySDR 0.7 from the 20.04 packages.

PE1NUT, 2021-01-01:
As usual - after looking for too many hours, I seem to have found the correct values just after posting a question.
Antenna: RX2
Clock Source: Internal
Driver arguments: driver=uhd

closed 2021-01-01 by PE1NUT

mpbraendli, 2021-01-03:
> As usual - after looking for too many hours, I seem to have found the correct values just after posting a question.

That happens a lot to me too :-)

Maybe `SoapySDRUtil --probe` would have helped you.