AlbrechtL/welle.io #618: Synchronization loss

vdwjeremy, 2021-01-03:
hello,
I loose frenquently synchronization with error "Lost coase sync" and "SyncOnPhase failed",
When that happens the frequency correction varies widely and the audio is lost for a few seconds.

https://user-images.githubusercontent.com/32203861/103485381-1d12d880-4df6-11eb-90ba-25354df09afd.mp4

SDR: Nooelec NESDR SMArt v4 (RTL2832U based)
Welle version: 2.3-dev build 2020-01-02
OS: Windows 10

Coarse corrector: CorrelatePRS (deactivated or others correctors are less effective)

The signal seems strong so I'm not sure what's happening here...
Any idea ?

andimik, 2021-01-04:
Could you try if the same problems occur also when using rtl_sdr?

In my case it was helpful.

vdwjeremy, 2021-01-04:
How do you do that ?

vdwjeremy, 2021-01-04:
To rule out hardware or reception issues, i tried out [https://www.ukwtv.de/cms/index.php?option=com_content&view=article&id=281&Itemid=550&jsmallfib=1&dir=JSROOT/dabplayer](https://www.ukwtv.de/cms/index.php?option=com_content&view=article&id=281&Itemid=550&jsmallfib=1&dir=JSROOT/dabplayer) (after switching drivers) and it works fine.
This player uses Realtek driver RTKDAB.dll directly, so not open source unfortunately.

beermad, 2021-02-12:
A thought which may, or may not, be helpful.

Do the sync losses occur at regular times? The multiplex I mainly record (using welle-cli) has certain times when channels' bitrates change (usually because extra channels are  added to or removed from the mux). This throws Welle off
[welleErrors.log](https://github.com/AlbrechtL/welle.io/files/5970707/welleErrors.log)

, so I have to re-spawn the process to continue my recording. These changes happen at the same time each day (for example at 0900 and 0945, as there's an extra channel between those times).

It could be that the other player OP has tried handles bitrate changes more cleanly than Welle?

I've attached a segment of log showing the messages received when a mux reconfiguration occurs.

andimik, 2021-02-12:
@beermad

Could you open a new issue?

According to your report this looks like a different story.

beermad, 2021-02-12:
> @beermad
> 
> Could you open a new issue?
> 
> According to your report this looks like a different story.
> 
> And please attach the full log as your file only shows the tail.

Will do. I'll gather some comprehensive logging and post an issue in a couple of days when I've caught the right events.

AlbrechtL, 2021-03-27:
@vdwjeremy Thanks for reporting. Which CPU your are using? 

@beermad Is your issue the same like https://github.com/AlbrechtL/welle.io/issues/587?

beermad, 2021-03-27:
> @beermad Is your issue the same like #587?

Hi,
Looking at the log attached to that issue, it's showing the same messages I get when my muxes change config. So they're probably both the same.
Thanks.

AlbrechtL, 2021-05-22:
OK. I marked it as duplicated.

closed 2021-05-22 by AlbrechtL