AlbrechtL/welle.io #619: welle.io running on Windows 7 (64 bit)

BrianG61UK, 2021-01-08:
I can scan and get a list of available stations and I can click on one and it plays.
But that's it if I want to play another I have to quit the program and run it again.
It will receive another but without sound.
I'm sure earlier versions I tried did this properly, though they were not as stable as 2.2.

andimik, 2021-01-08:
Please mention the two programs.

Maybe the sampling rates are different (48 or 32 kHz). See https://github.com/AlbrechtL/welle.io/issues/573

BrianG61UK, 2021-01-11:
It's working better today, or I was just unlucky in which stations I chose yesterday.
Changing either to or away from things like "LBC News" which have 32kHz sample rate audio seems to be the main problem.

orryverducci, 2021-01-15:
@BrianG61UK the problem you're having is definitely the sample rate bug reported in #573.

If you tune to a station that has a different sample rate to the first one you play, welle.io stops outputting audio. The UK national multiplexes have a mix of sample rates so you quickly run in to this problem. If you tune between stations that are all the same sample rate (e.g. the BBC stations) it should continue playing.

This has been fixed in the code but has not yet made it in to the latest release. You can download nightly builds which have this fixed from [https://bintray.com/albrechtl/welle.io/welle.io_nightly].

To answer your question, yes LBC News is really 32kHz. The AAC codec used by the DAB+ stations support a range of sample rates between 16kHz and 48kHz. Global have chosen to use 32kHz for their DAB+ stations as it gives a small sound quality improvement at the low bitrates they're using while maintaining a similar audio bandwidth to FM radio.

BrianG61UK, 2021-01-15:
Thank you thank you.
I had success with the bottom file at https://bintray.com/albrechtl/welle.io/welle.io_nightly#files
Bizarrely, this file  (20210102_fbcb9f6_Windows_welle-io-setup.exe) downloads renamed to  20210102_fbcb9f6_Windows_welle-io-setup.zip in Firefox (but not in Chrome) but I figured it out after I found nothing would open this zip file.

closed 2021-01-15 by BrianG61UK

BrianG61UK, 2021-01-16:
> Global have chosen to use 32kHz for their DAB+ stations as it gives a small sound quality improvement at the low bitrates

Actually to my ears this is not an improvement.
The artificial HE part of HE-AAC works nicely for my ears artificially generating the top octave 12kHz-24kHz but with it generating 8kHz-16kHz it definitely sounds bad, like an MP3 where the bitrate is too low, there are audible nasty high frequency artefacts that grate and make it unpleasant to listen to.

AlbrechtL, 2021-05-22:
Please try again the latest nightly build.