AlbrechtL/welle.io #620: welle-cli convert to AAC

antonioag95, 2021-01-11:
Would it be possible to encode the station to AAC instead of MP3? I figured out how to increase the current MP3 bitrate from 128 to 320 but encoding to AAC would need another library, not "lame". As of almost every device is now capable of playing AAC that would mean less bandwith with more audio quality. It would be awesome if metadata could be added (e.g. the dynamic label as song title or the picture received as image). Thank you very much!

mpbraendli, 2021-01-18:
Hi, it's certainly doable, but I find the advantage/effort ratio to be rather low. mp3 is supported nearly everywhere, and the quality is mostly given by the source stream, not by the codec used to carry from welle-cli to the client.
I don't know how easy it would be to add the dynamic label to the mp3 metadata. Not sure if it has to be ICY-Text or an ID3 tag.

antonioag95, 2021-01-18:
Hi, thank you for your answer.
I'm aware about the quality - which is given from the source - but it also true that a 96 kBit/s source stream in AAC sounds way better than the re-encoded 128 kbit/s stream. In order to keep the same AAC audio quality in MP3 we need to increase it to 192 kbit/s or more but it is translated in more bandwidth/traffic to exchange between server and client, that's why I thought it was a good idea. Maybe they can be left both and let the user chose the preferred encoder and bandwidth :) 
As per labels, I think that ICY is only used for Icecast streams, I might be wrong tho. 
Thank you again.

marcusmueller, 2021-03-03:
@antonioag95 just to understand this: You want to re-encode the MPEG 4 HE-AACv2 bitstream to a different AAC derivative, right, not just get the native MPEG 4 audio bitstream as it is?

antonioag95, 2021-03-03:
@marcusmueller affirmative. Well, actually I wish we could use the native MPEG4 as it is but due to its not standard 960 samples per frame it causes some issues on standard players - kinda slowed down audio effect - so it needs to be first at least re-encoded to 1024 samples per frame AAC.

AlbrechtL, 2021-05-22:
Pull request are welcome! In the meantime I'm closing this issue because it is not our priority to implement it.

closed 2021-05-22 by AlbrechtL