AlbrechtL/welle.io #621: MOT Slideshow not updating

vk1kcm, 2021-01-23:
When listening to some Australian DAB+ stations the MOT Slideshow doesn't update when a new file is downloaded.  It looks like they are using the same filename for each file but I don't know if that would cause this issue.  If I remove and re-add the MOT Slideshow the latest slide appears.  See the attached screenshot.
[image]

Jasoroony, 2021-01-24:
Change MotView.qml to have a few extra lines (turn cache off and force a reload) and recompile:

```
                Image {
                    id: motImage

                    cache: false
                    fillMode: Image.PreserveAspectFit
                    Layout.fillWidth: true
                    Layout.fillHeight: true

                    Connections{
                        target: guiHelper
                        onMotChanged:{
                            latestPicture.pictureName = pictureName

                            // Display slide show only of latest tab is select
                            if(bar.currentIndex === 0) {
                                motImage.source = "image://SLS/empty"
                                motImage.source = "image://SLS/" + pictureName
                            }
```

andimik, 2021-01-24:
If this works, please send a Pull Request.

nickpiggott, 2021-02-24:
[TS 101 499](https://www.etsi.org/deliver/etsi_ts/101400_101499/101499/03.01.01_60/ts_101499v030101p.pdf) says (section 6.2)

"The ContentName shall be changed for each new slide, and the value shall never be reused in reference to different
content."

So the Australian implementation is not to standard. The ContentName (Picture Name) should be different for each slide that is different. This is to aid caching on the receiver.

AlbrechtL, 2021-05-22:
@vk1kcm Thanks for reporting. Indeed it is a non-Standard behavior. Feel free to inform the radio station. You can also refer to this issue if you like. 

Because it is no welle bug. I'm closing this issue.

closed 2021-05-22 by AlbrechtL