AlbrechtL/welle.io #623: Document to open PR only for the next branch

AlbrechtL, 2021-01-26:
Details are here https://github.com/AlbrechtL/welle.io/pull/622#issuecomment-767106907

andimik, 2021-01-26:
But you don't want to set the *next* branch to default, right?

marcusmueller, 2021-03-03:
Recommendation: You can add PR templates to your repo, allowing you to predefine the text people see, including  checkboxes like:

- [ ] I've made sure code is properly indented / formatted
- [ ] If this is anything but a bugfix that specifically targets a released version, I'm doing a PR against `next`, not any other branch.
- [ ] My hovercraft is full of eels

AlbrechtL, 2021-05-22:
Thanks for the hint! Now it is just a matter of doing.

> But you don't want to set the next branch to default, right?

Yes, because the next branch is the development branch and sometimes the code is broken.

AlbrechtL, 2026-10-22:
Closed because the next branch will be gone soon (see https://github.com/AlbrechtL/welle.io/issues/954).

closed 2026-10-22 by AlbrechtL