AlbrechtL/welle.io #625: CAN'T FIND ANY CHANNEL

aidandorc, 2021-02-20:
Hi..I'm new here.
I've just finished to compile welle.io on my linux mint box (20.1), same environment of an ubuntu distro.
Everything was good for the gui and cli programs.
I'm using an RTL2838 usb stick with its antenna plugged-in and it is correctly detected.
But when I search for Dab channels on welle gui, anything is found. No channels at all. In the Service Detail panel the stick is detected but all other settings are red (Frame sync, Fic crc......)
Am I missing something?
Can anyone help please? 
Thanks in advance...
Bye

andimik, 2021-02-20:
Well, you don't mention where you are living. Do you have a DAB car radio or any other small DAB radio to test if you can receive it?

How strong do you expect the signal? Maybe you should try to use the manual gain or use an external (roof) antenna.

aidandorc, 2021-02-20:
I live in Milan, Italy. A big city. No problems of signal here....
No dab car radio, I'm using my laptop with the usb stick plugged in.
What do you mean with manual gain? How to change this setting?
thx and bye

andimik, 2021-02-21:
Use the expert settings and scan on 12B (RAI), 12C (DAB Italia) and 12D (EuroDAB).

Please post further info (from the console, screenshot from 12C spectrum diagram, etc.).

aidandorc, 2021-02-21:
[image]

it doesn't work

andimik, 2021-02-21:
Does the same stick works with Qt-DAB?
Have you tried the manual gain?
Tried another antenna?

I fear it's not related to Welle.io.

aidandorc, 2021-02-21:
[image]
yes...it's an antenna problem.....I'm working on it.....
thx

gvanem, 2021-02-21:
With a 96 kBit/s stream you need at least 13 dB SNR in my experience.
Try a "Talk Channel" with e.g. a 48 kBit/s stream. Then maybe a SNR of 9 or 10 dB is enough.

aidandorc, 2021-02-21:
do you have any suggestions to improve antenna performances?
I'm using the default antenna of the usb stick kit..if I touch it with my hands the signal is ok, If I leave it the signal goes down....

andimik, 2021-02-21:
I fear you are using a wrong antenna. Yours seems to be optimized for UHF (TV).

Don't you have another antenna?

In Milan, 12B and 12C have a transmitter in the town. With a better antenna you might even receive up to three multiplexes from Switzerland!

andimik, 2021-02-21:
@gvanem 
I am not a technician, but in my opinion the Bitrate has nothing to do with the SNR. I fear you are mixing it up with the protection level.

In Milan, nearly all services use Protection Level 3A, and the only KC1TEST uses 2A.

gvanem, 2021-02-21:
@andimik I wrote it is vice versa; SNR has everything to do with how many Bit/s you can get through. 
No matter how good the Viterbi coding and all that is. If not <img src="https://latex.codecogs.com/gif.latex?\inline&space;\small&space;SNR" title="\small SNR" /> == carrier(s)-level to noise, what does SNR really 
mean in Welle-IO?

Any communication channel requires a certain Signal to Noise ratio for a certain rate of information, no?
Imagine a shortwave receiver with a demodulation-bandwidth of 1 Hz. How many *Dits* and *Dahs* of Morse-code (CW) could get through per second? Pretty close to 1 Dit or *Dah* per second. I won't go into detail on *Channel capacity*, *Shannon*, <img src="https://latex.codecogs.com/gif.latex?\inline&space;\small&space;E_b/N_0" title="\small E_b/N_0" /> etc. I'm not sure I remember much of it.

aidandorc, 2021-02-21:
> I fear you are using a wrong antenna. Yours seems to be optimized for UHF (TV).
> 
> Don't you have another antenna?
> 
> In Milan, 12B and 12C have a transmitter in the town. With a better antenna you might even receive up to three multiplexes from Switzerland!

Please suggest me the right model. Consider that by stick side there is a micro connector, it's not the standard antenna connector.....

gvanem, 2021-02-21:
Split the coax-cable with a sharp knife and use the 2 parts as a dipole. 
Like here: https://www.youtube.com/watch?v=tafy51yUzQo&ab_channel=KevinLoughin
Time 03:30.

marcusmueller, 2021-03-03:
Honest, and well-intending question: Is the software issue tracker the right place to discuss reception problems, and antenna designs?

closed 2021-03-04 by aidandorc

AlbrechtL, 2021-05-22:
Because we don't have forum I have no problem to discuss reception issues.

I'm using an old (> 40 years) UHF TV Yagi antenna to receive the DAB signals.