AlbrechtL/welle.io #633: ALSA SUPPORT ISSUE

aidandorc, 2021-02-24:
Hi
I'm trying to use welle-cli but I think a problem of audio exists.
Here the output:
sudo ./welle-cli -c 12C -p "Deejay"
Hello this is welle-cli d34b3fc
InputFactory:Input device:auto
Airspy: Open airspy
Airspy: airpsy_open () failed: AIRSPY_ERROR_NOT_FOUND(-5)
RTL_SDR: Open rtl-sdr
RTL_SDR: Found 1 devices. Uses the first working one
Found Rafael Micro R820T tuner
RTL_SDR:  Opening rtl-sdr device0
[R82XX] PLL not locked!
RTL_SDR:  Supported gain values29
RTL_SDR:  gain 49.6
RTL_SDR:  gain 48
RTL_SDR:  gain 44.5
RTL_SDR:  gain 43.9
RTL_SDR:  gain 43.4
RTL_SDR:  gain 42.1
RTL_SDR:  gain 40.2
RTL_SDR:  gain 38.6
RTL_SDR:  gain 37.2
RTL_SDR:  gain 36.4
RTL_SDR:  gain 33.8
RTL_SDR:  gain 32.8
RTL_SDR:  gain 29.7
RTL_SDR:  gain 28
RTL_SDR:  gain 25.4
RTL_SDR:  gain 22.9
RTL_SDR:  gain 20.7
RTL_SDR:  gain 19.7
RTL_SDR:  gain 16.6
RTL_SDR:  gain 15.7
RTL_SDR:  gain 14.4
RTL_SDR:  gain 12.5
RTL_SDR:  gain 8.7
RTL_SDR:  gain 7.7
RTL_SDR:  gain 3.7
RTL_SDR:  gain 2.7
RTL_SDR:  gain 1.4
RTL_SDR:  gain 0.9
RTL_SDR:  gain 0
RTL_SDR: Start rtlsdr_read_async_wrapper() thread
Allocating 15 zero-copy buffers
OFDM-processor:restart
Wait for sync
ofdm-processor: Lost coarse sync (coarseCorrector: 0; fineCorrector: 0)
ofdm-processor: SyncOnPhase failed
ofdm-processor: SyncOnEndNull failed
Ensemble name id: 5009
ofdm-processor: Found sync (coarseCorrector: 0; fineCorrector: 0 after 17 frames)
New Service: 0x5215
New Service: 0x5219
New Service: 0x5264
New Service: 0x5286
New Service: 0x5281
New Service: 0x5284
New Service: 0x51cc
New Service: 0x5210
New Service: 0x5214
New Service: 0x5233
New Service: 0x5245
New Service: 0x5289
New Service: 0x5283
New Service: 0x52cc
New Service: 0x5282
New Service: 0x5280
New Service: 0x5288
New Service: 0x5285
New Service: 0x5287
New Service: 0xe0500001
Ensemble label: * DAB Italia *
Wait for service list
{"TII":{"comb":3,"delay":132,"delay_km":19.3359375,"error":282.0,"pattern":1}}
Ensemble label: * DAB Italia *
{"TII":{"comb":3,"delay":221,"delay_km":32.373046875,"error":246.0,"pattern":1}}
Ensemble label: * DAB Italia *
Ensemble label: * DAB Italia *
{"TII":{"comb":3,"delay":93,"delay_km":13.623046875,"error":161.0,"pattern":1}}
Nothing to do, not ALSA support.
OFDM-processor: closing down
OFDM-decoder:closing down now
----------------
It says NOT ALSA SUPPORT but alsa is installed on the system (notebook with a Mint 20.1).
Any suggestions appreciated....
Thx Bye

mpbraendli, 2021-02-24:
Hello, you probably didn't have the ALSA development package installed when you compiled welle.io. Make sure `libasound2-dev` is installed.

aidandorc, 2021-02-24:
> Hello, you probably didn't have the ALSA development package installed when you compiled welle.io. Make sure `libasound2-dev` is installed.

ok....libasound2-dev installed, then re-compiled but same error....
some other components to add?

mpbraendli, 2021-02-25:
I don't think so. I hope you re-ran cmake too, not just `make`. Ideally delete the build folder first.

aidandorc, 2021-02-25:
> I don't think so. I hope you re-ran cmake too, not just `make`. Ideally delete the build folder first.

Ok, thanks.
I deleted all old build folders, re ran from start..but same problem and message.
I'm using to compile the suggested graphic app QT Creator. From menu you can make qmake not cmake.
Could it be the problem?
thx

mpbraendli, 2021-02-25:
I don't know the qmake build welle enough. You should build it with cmake instead, have a look at the readme first. I'm quite sure that's the issue.

andimik, 2021-02-25:
@aidandorc

As you just want to build welle-cli the Qt Creator is not necessary.

Please follow the instructions in the readme (depending on your hardware wishes) and use `cmake`.

aidandorc, 2021-02-26:
> @aidandorc
> 
> As you just want to build welle-cli the Qt Creator is not necessary.
> 
> Please follow the instructions in the readme (depending on your hardware wishes) and use `cmake`.

Yes it's true...now it works.
Thanks

closed 2021-03-03 by AlbrechtL