AlbrechtL/welle.io #648: Please add topics to the repository

mendel5, 2021-03-26:
It would be great if you could add the following Github topics to this repository:

- [https://github.com/topics/sdr](https://github.com/topics/sdr)
- [https://github.com/topics/rtl-sdr](https://github.com/topics/rtl-sdr)
- [https://github.com/topics/dab](https://github.com/topics/dab)
- [https://github.com/topics/dabplus](https://github.com/topics/dabplus)

Thanks!

andimik, 2021-03-27:
What is the benefit?

The 2nd shows 219 hits. The majority has nothing to do with DAB.

mendel5, 2021-03-27:
The benefit is that these topics allow people to find the project/ repo based on the general subject even if they don't know the project name.

With the second link you probably mean [https://github.com/topics/rtl-sdr](https://github.com/topics/rtl-sdr), right? Of course SDR (software defined radio) has something to do with welle.io, it also says that in the description:
> A DAB/DAB+ Software Defined Radio (SDR)

andimik, 2021-03-28:
Under this circumstances (the keywords are already in the Readme) I'd like to propose closing the issue (which is not an issue in fact).

AlbrechtL, 2021-05-22:
Done! Thanks for the hint!

closed 2021-05-22 by AlbrechtL

mendel5, 2021-05-22:
Sure!