AlbrechtL/welle.io #653: Feature Request: Radio audio Recording support

nanderer, 2021-05-15:
Hi there,
i just installed welle.io to test my new rtl-sdr, in genral it works grate with my bad indoor antenna, much better then listening to FM using Gqrx (all noisy), but there is one function i am missing: recording

i would like to be able to recorde at least the one i am hearing right now, even better, multiple channels at once, i can record and replay the raw stuff but to record the audio now i had to record the output stream of my pulseaudio, so please build in a red recording button that you can just click and then get some .wav or even better .opus or .flac file tagged with metadata and recording start time. 

Thanks a lot and keep up the good work

nanderer, 2021-05-15:
maybe even better, just save the aac-stream without reencoding...

AlbrechtL, 2021-05-22:
You can use the command line option `--dump-file` for file dumping. But you cannot play this file with a Standard audio player because DAB+ is using different AAC parameters compared to the Standard.

The easiest would be to implement a WAV recording feature. But welle.io is a radio listen program so I don't see a benefit for recording of the radio program. If you want to record your favorite radio program you can just record the MP3 internet live stream.

closed 2021-05-22 by AlbrechtL