AlbrechtL/welle.io #656: feature request: setting to go more easy on rtl_tcp

nanderer, 2021-05-17:
hi,

i have just tested rtl_tcp on my router, which is a grate thing to have the antenna in a fixed location and access the sdr using wifi, but it of cause needs to configured to use less bandwidth, on SDR# it was just reducing the bandwith setting (sample rate) to under 2.048 MSPS, on gqrx i had to decrease it even more to 960000 ksps, looks like it is not implemented as efficient and just gets the same data more then one time at once for different functions instead of doing local multiplexing. 

But guess what i didn't get to work: welle.io.

so it would be nice if you could add an option that gets more easy on the rtl_tcp interface...

Thanks a lot in advance and keep up the good work

andimik, 2021-05-17:
Please don't try to decrease the sampling rate for DAB.

Use a cable instead, if your wifi is not good enough.

gvanem, 2021-05-18:
The sample-rate is defined by a `#define INPUT_RATE 2048000`.
I experimented with setting it to `2000000` which caused no signal-lock. Why is it so critical?

andimik, 2021-05-18:
Because I warned you. That's why 😁

AlbrechtL, 2021-05-22:
I'm glad that you are asking.

The minimal sample rate is physically and mathematically limited by the [Nyquist–Shannon sampling theorem](https://en.wikipedia.org/wiki/Nyquist%E2%80%93Shannon_sampling_theorem). DAB and DAB+ is using a 1.536 MHz wide bandwidth. Because we are using I/Q sampling the absolutely minimum sample rate is  `1536000`. But practically this will not work because we need some frequency space around the signal e.g. for the SNR calculation. The sample rate `2048000` is used because the FFT is most efficient for counts for exponent by 2.

Did I answer you question? 

If you don't understand from what I'm talking about I recommend some free lectures in signal processing e.g. here https://ocw.mit.edu/resources/res-6-008-digital-signal-processing-spring-2011/.

closed 2021-05-22 by AlbrechtL

nanderer, 2021-06-14:
i was not talking about sampling rate but about the radio bandwith (like in khz). the sampling rate of cause sould stay the same but i compared the use of different 

[image]
Gqrx 

[image]

regards

andimik, 2021-06-15:
@nanderer 

Can't you reopen it here?

So you want to decrease the bandwidth?

Why don't you start rtl_tcp with the proper parameter?

https://github.com/hayguen/extio_rtl_tcp/releases

https://www.rtl-sdr.com/new-extio-for-rtl_tcp-control-r820t-bandwidth-decimation-auto-reconnect/

Of course it's available for Linux as well.

Do you have a special case where you need to decrease the bandwidth? Neighbor channel reduction?