AlbrechtL/welle.io #657: [cli] proposal archive

d2009, 2021-05-24:
Can the following be built into cli: display of the last 3 or 4 slideshow images, as well as the last radio texts about 10 min?

d2009, 2021-10-03:
Is there anything new about the archive?

mpbraendli, 2021-10-20:
Sorry I won't have time to look into that.

AlbrechtL, 2022-01-06:
I'm closing this topic for know because of inactivity. Anyway: pull requests are welcome!

closed 2022-01-06 by AlbrechtL

andimik, 2022-01-07:
For such a reason you might want to try Qt-DAB. All slides (including SPI) are saved to `/tmp/qt-pictures/` (have not tried, which path it is under Windows) and optionally you can save the DL into a file.

aerogus, 2022-09-12:
I started a simplified fork of welle-cli with aim to archive and historize data (mot, dls ...) and audio
I'd be happy to propose some PR to the initial welle.io project if some people with the same needs are interested in.
https://github.com/aerogus/welle-cli

mpbraendli, 2022-09-12:
Bonjour @aerogus et merci pour ton message!

I have the impression you want to take welle-cli into a new direction with your fork, with the aim to log ensembles locally to files. Would it make sense to add include your changes into the welle.io repository under a new binary, for example `welle-ensemble-recorder` ? That way, we don't risk to break things in welle-ci, but we can still work together on the same repository.

@AlbrechtL would you accept pull requests with that aim?

aerogus, 2022-09-17:
Thanks for your feedbacks.
ok, I'll will translate to english all the messages.
The audio output is raw pcm because I didn't know how to export the aac frame (and I'd want to ! help appreciated ;)).
The webserver and lot of stuff had been removed just for my better unterstanding and learning the welle.io codebase.
I hope in the future to be able to propose some pull requests into the main project.