AlbrechtL/welle.io #660: RTL_TCP Support for Android Version

pathfinder1803, 2021-06-09:
Hello,
I would like to use Welle.io at my Android Smartphone while sharing the screen to my Car via Androidauto which is already supported. The problem is that I can't plug in the RLT SDR dongle not at the same time. So Androidauto Support doesn't make sens in my point of view.
I would suggest to implement RTL_TCP Support for the android app as a second option. This way sharing the signal by a second device would be possible and the usage for cars is possible.
Thanks in advanced.

tenzap, 2021-06-10:
Just an idea, don't know if this would fit, but what about using a usb hub ?

pathfinder1803, 2021-06-10:
Good Idea. I already tried this.
The problem is that the phone can either be USB-Host (for USB Dongle) or USB-Client (for the car via OTG). This way the hub would work for the car radio but this doesn't help at all.

AlbrechtL, 2021-09-10:
Please check the next branch. There rtl_tcp is supported even for Android.

closed 2021-09-10 by AlbrechtL

pathfinder1803, 2021-09-16:
Can you provide this branch as a released APK?

AlbrechtL, 2022-01-06:
Can you find APKs here: https://welle-io-nightlies.albrechtloh.de/

pathfinder1803, 2022-01-08:
Thanks. It works great!