AlbrechtL/welle.io #663: Tune in via service id with welle-cli

white-gecko, 2021-06-25:
When I want to tune to the "Dlf" (Deutschlandfunk) welle-cli tunes to "Dlf Nova" as it does a substring matching

https://github.com/AlbrechtL/welle.io/blob/66675c9a4160c644a5ade5bf7a0cd0647647f33c/src/welle-cli/welle-cli.cpp#L621

Is there any chance to tune to a programm with its id? e.g. `0xd210` in case of Dlf.

I coulnd't find out which filed holds the service id on the service object, so I couldn't make a pull-request, sorry.

Nice project Albrecht!

mpbraendli, 2021-06-28:
How about we try to do an exact match, and do the prefix match only if that fails?

white-gecko, 2021-06-28:
I think that is a good idea for tuning via station name. But an additional parameter for the id could help as well.

AlbrechtL, 2022-01-06:
@mpbraendli Can we close this topic?

But I always say pull request are welcome!

white-gecko, 2022-01-06:
Actually I have just these days setup a raspi with welle-cli and I want to listen to dlf, so I will need to submit a pullrequest when I get to the point ...

thaenig, 2023-10-30:
Yesterday I did my first steps ever with welle-cli und stumbled upon the issue with Dlf too. What worked as station name was "Dlf  " - two spaces after the actual name and everything within double quotes.

mpbraendli, 2024-10-09:
Closing due to lack of activity.

closed 2024-10-09 by mpbraendli