AlbrechtL/welle.io #664: Volume issue in 2.3

Annunaki, 2021-07-10:
Output volume is reset to 100% when tuning to different stations, however, the Volume Slider remains at where it was set.

AlbrechtL, 2021-12-28:
Thanks for reporting!

AlbrechtL, 2022-08-21:
Is this bug is till occurring?

SergeantSerk, 2022-08-22:
Interestingly, I stumbled on this issue right now and was about to start an issue. Yes it is still occurring.

Switching stations resets the volume to 100% but the slider remains in the same position.

AlbrechtL, 2022-12-16:
Thanks for testing! Bug fix pull request are welcome!

mdmitry, 2024-03-16:
Still not fixed in 2024

AlbrechtL, 2024-10-08:
The described issue I can't reproduce. But I found another issue that the volume was always 100% after mute that is fixed here https://github.com/AlbrechtL/welle.io/commit/86a6fdad9b9eb44b47bf74564cacdec6a5d8ccd4.

Can you try it with this version (https://welle-io-nightlies.albrechtloh.de/20241008_86a6fdad_Windows_welle-io-setup_x64.exe), please?

closed 2024-10-08 by AlbrechtL