AlbrechtL/welle.io #665: Slides don't update

Observerist, 2021-07-11:
v2.3 latest nightly 2021/06/07 for OSX

Paris, France : 
Slides are not being updated, same behavior even on major stations supposedly following rules. 
Excepted are stations without their SLS linked to a playlist, where the slides simply run on a timer. 

Win-based DAB Player does update slides correctly.

tenzap, 2021-07-11:
duplicate of #547 ?

Observerist, 2021-07-11:
Can't verify the #581 part mentioned in #547... only the non-updating SLS. 
Have to switch to another station and back again ( inside or outside the Mux ) to get the actual slide in the MOT window.

BTW, what happened to the path indication seen in prior releases ? No more ? 
Not that I could ever do something with it, lacking any instructions for its use.

AlbrechtL, 2021-12-28:
Do you have a RAW I/Q data file?

Observerist, 2022-01-02:
What duration / filesize do you need ? 
Where do I send / deposit the IQ file ?
With a link to a cloud server, like wetransfer, google drive or similar ?

andimik, 2022-01-02:
Well, let me check with M1 and M2 which I can receive via Eutelsat 5W.

Observerist, 2022-01-02:
There are no problem cases with locked slides on M1 & M2.
Some cases are on the Parisian 6D, not sure that you can get these stations via Fransat on 5ºW.

The following stations are the ones having problems with the refresh of their slides :
MBS
LOVELY
Not Life Threatening, just annoying for some good stations.

Don't have access right now to a Win PC to countercheck with the DAB Player

andimik, 2022-01-02:
If you have installed eti-cmdline from https://github.com/JvanKatwijk/eti-stuff you could share an ETI file. You could compress it as zip or 7z in order to reduce the file size.

Observerist, 2022-01-04:
More Info :
Tested Paris 6D stations MBS & LOVELY using DAB Player on Windows.
No problems with the slides, they are refreshed the nominal way.

AlbrechtL, 2022-01-06:
Since welle.io uses the MOT code from [dablin](https://github.com/Opendigitalradio/dablin) maybe @basicmaster has an idea?

basicmaster, 2022-01-06:
Difficult to say without having a look at the current transmission. I have two dumps from 2018 when MBS was transmitted on 11A as part of "RNT Paris 6". But it has the same slide in both dumps and also the label doesn't change. In that case it is already part of the transmission and hence a bug on transmission side. The current situation seems to be different.

@Observerist: Could you provide MP4 recordings of both stations (a few minutes), made with the DAB Player on Windows? This would be sufficient, if an ETI dump is not possible. Does Dynamic Label work at the same time, both in welle.io and DAB Player?

Observerist, 2022-01-07:
@AlbrechtL  Happy to know that you are OK. That's what matters. Your creation seems to be well taken care of.

@basicmaster  The DLS is showing up on both platforms fully in sync. And the Slideshow window in the DAB Player is refreshing its content every couple of seconds. While welle is always stuck after the first slide.

Made the two mp4 audio recs as requested. Did also a bigger raw recording of Paris 6D.
Would you like to retrieve these files via a deposit on wetransfer ? 
Or do you have another favorite ? Private mail needed ?

basicmaster, 2022-01-07:
Great! WeTransfer would be OK for me. If necessary, you can find my e-mail address on my website (see my profile) as part of the "Impressum".

basicmaster, 2022-01-07:
Thanks for the dumps. The two MP4 dumps actually look quite normal, so I wanted to have a look at the raw dump. Unfortunately it is too short to contain a slide change. Could make a dump of e.g. 5min which also contains from one song to another one?

basicmaster, 2022-01-08:
Thanks for the other dumps. With the first new dump and "LOVELY" I was able to check how DABlin behaves here: Upon the slide change, the new slide is shown.

[image]

With the new slide, the ContentName stays the same while the Transport ID is incremented:

[image]

DABlin just checks the Transport ID and ignores (but shows) the other parameters, so it shows the changed slide. Therefore also welle.io should get the changed slide.

> The two MP4 dumps actually look quite normal,

Actually, this is not the case: It turns out that according to [ETSI TR 101 499](https://www.etsi.org/deliver/etsi_ts/101400_101499/101499/03.01.01_60/ts_101499v030101p.pdf) § 6.2.2 the ContentName *shall* be changed for a new slide. And a displayed slide *shall not* be updated. So in the end welle.io behaves as defined.

@AlbrechtL assuming welle.io displays a new slide only if the ContentName is different, the behaviour could be modified to compare the Transport ID, I guess...

AlbrechtL, 2022-02-16:
@basicmaster Thanks for the analyzes!

Before we fix a broadcaster issue I would like to get a statement from the broadcast provider of MBS and LOVELY.

@Observerist Can you write to MBS and LOVELY with reference to this issue?

AlbrechtL, 2024-10-08:
Closed because of inactivity.

closed 2024-10-08 by AlbrechtL