AlbrechtL/welle.io #666: Feature request: Save playlist of songs listened to or of station

spongioblast, 2021-07-20:
Would be fantastic to have an option to save a playlist of the radios listened to via webinterface or simply save the playlist of a specific channel. (Artist, Song, Album). Then I would no longer have to soundhound them to make sure I wont forget the song. Thanks.

andimik, 2021-07-20:
There is no playlist. You probably mean DLS or even DL+.

spongioblast, 2021-07-21:
> There is no playlist. You probably mean DLS or even DL+.
Yes, that's what I mean. Would be perfect if the DLS info could be saved for a station.

spongioblast, 2021-07-21:
JvanKatwijk at Qt-DAB quickly created a DLS log option (https://github.com/JvanKatwijk/qt-dab/issues/223). Hence I believe it's no longer needed fore welle.io. With the purpose of welle.io being a Radio it would be great to have #602 implemented. This ticket can be closed from my side. Thanks!

closed 2021-12-28 by AlbrechtL