AlbrechtL/welle.io #671: Set raspberry pi's clock using DAB Time Signal

Kone1211, 2021-07-30:
Hello Fellows,

I am currently working on a car head solution with OpenAuto Pro. Its Operating System is Raspberry Pi OS.

Problem:

Due to the missing Real-Time-Clock (RTC) on the Raspi Breadboard, you have to buy on of these separately.

I've done this in order to have a working time display in my car. Turns out these RTC's arent very accurate. Every Time when I boot my OpenAuto-System, I'm gaining or loosing some seconds.

As a sidenote: The Raspberrypi is not connected to the Internet.

Possible Solution?:

Is it somehow possible to set the Time of the Raspberry Pi using the time provided by the DAB Signal?

Unfortunately I have absolutely no Idea how to write such a script/programm.

The only thing I can provide that I know you can set the time on the pi using this command: `sudo date --set '2021-07-30 18:29:35'` maybe someone can write something witch pulls the time from welle.io and paste it into this script at programm start.

Thanks in Advance

Kone

andimik, 2021-07-30:
Duplicate of https://github.com/AlbrechtL/welle.io/issues/654

andimik, 2021-07-30:
I'm using a simple GPS mouse for that.

https://github.com/hayguen/fmlist_scan for example has a solution.

Kone1211, 2021-07-31:
Thanks @andimik, I did not look carefully enough:-(

Anyway, your solution with the GPS Mouse sounds very straight voward. 

Thanks for your help!

I'm closing this issue now

closed 2021-07-31 by Kone1211