AlbrechtL/welle.io #673: Alter frequency / Add channel in config or command line

mattaperkins, 2021-08-22:
Hi, 
 Is there a way to alter the centre frequency or add a new channel on the command line or in a config file.  Other then altering the source code. I have not found a way.  
 
 I would like to use another frequency in the Amateur Radio 70cm band.

andimik, 2021-08-22:
https://github.com/AlbrechtL/welle.io/blob/f97165367458cc6b079dfe6b9113405fecfb28be/src/various/channels.cpp is the code.

Do you want to modify it for cable DAB radio?

mattaperkins, 2021-08-22:
Yes I have modified that part of the code to do some experimentation to use welle.io on Amateur Radio  frequency's. However i could not find anything that would allow perhaps to set the frequency on the command line or type it in directly.  It appears to work however I have not setup my transmitter yet to test the decoding. 
Matt

andimik, 2021-08-23:
Are you sure you will find DAB on amateur radio frequencies?

If you already modified this code you can select the new channel in the expert view.

mattaperkins, 2021-08-23:
Yes im the one going to be broadcasting it. :)  I wanted an easy to use good piece of software for other amateurs to receive the DAB+ broadcast. 

Kind regards
Matt

AlbrechtL, 2021-12-28:
Interesting application. How often is DAB broadcasted over HAM radio frequencies?

AlbrechtL, 2024-10-08:
Closed because of inactivity.

closed 2024-10-08 by AlbrechtL