AlbrechtL/welle.io #675: macOS: RTL_TCP_CLIENT:ADC overload. Maybe you are using a too high gain.

ambanmba, 2021-09-18:
macOS version gives this error when accessing device over TCP. Works fine when accessed locally via USB port. Other applications (e.g. CubicSDR) can access the RTL-TCP device without any problems.

On the server side the error is:
SoapyServerListener::accept([::ffff:192.168.1.152]:62460)
SoapyServerListener::handlerLoop() FAIL: SoapyRPCUnpacker::recv() FAIL: header word
SoapyServerListener::close()

ambanmba, 2021-09-20:
Just to add to this, I installed the Windows version of welle.io and it fully crashes and exits when trying to access trl-tcp

AlbrechtL, 2021-12-28:
I'm a little confused.
1. Do you talk about a rtl_tcp server or a Soapy server?
2. Which welle.io version are you using?
3. Did you enabled auto gain?

AlbrechtL, 2024-10-08:
I'm closing this issue because I'm not sure if it is still relevant. If it still occurs in the latest version, we can reopen it.

closed 2024-10-08 by AlbrechtL