AlbrechtL/welle.io #676: DAB+ webstream

peter-vanpoucke, 2021-09-25:
Might be I've missed this in documentation, but can as well ask. I could listen to a DAB stream using rtl_tcp and welle.io on Windows and welle-cli on Debian.
Just was wondering if there's a way to stream the DAB audio of a program (or a channel) over the network?

peter-vanpoucke, 2021-09-25:
OK, so they're described in `mux.json`, found it 😅. So maybe I can turn this in a new feature? Any plans on streaming the raw stream?

aerogus, 2021-09-27:
by raw stream you mean the he-aac from the eti ?
I'd also like to be able to extract it with welle-cli, and not only the decoded wav/pcm file, and not only the decoded/reencoded mp3 (which is a great feature).

AlbrechtL, 2021-12-28:
Which RAW stream do you mean? The pure RAW stream would be I/Q data e.g. via rtl_tcp which is supported.

Or do you mean ETI? We already had a input discussion here https://github.com/AlbrechtL/welle.io/issues/199 and here https://github.com/AlbrechtL/welle.io/issues/217.

Because of inactivity I'm closing this issue.

closed 2021-12-28 by AlbrechtL