AlbrechtL/welle.io #684: Dongle not recognized

markwinger, 2021-10-21:
I am attempting to us welle.io as part of openauto pro.  I can find no docs on how to configure this.  It appears that welle.io should auto recognize the dongle, but I get the error "No valid device found use Null device instead."  But I can find no config for setting the device.  How is this supposed to work.  (The dongle does work, I have successfully had gqrx tune to a station)

markwinger, 2021-10-21:
Running on Raspberry pi 4 .

Lsusb:
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 008: ID 08bb:2902 Texas Instruments PCM2902 Audio Codec
Bus 001 Device 007: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
Bus 001 Device 006: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 005: ID 8086:0808 Intel Corp.
Bus 001 Device 004: ID 1a40:0201 Terminus Technology Inc. FE 2.1 7-port Hub
Bus 001 Device 003: ID 0b05:190e ASUSTek Computer, Inc.
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

dmesg:
1808.204248] rc rc0: Realtek RTL2832U reference design as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3.4/rc/rc0
[ 1808.204434] rc rc0: lirc_dev: driver dvb_usb_rtl28xxu registered at minor = 0, raw IR receiver, no transmitter
[ 1808.204599] input: Realtek RTL2832U reference design as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3.4/rc/rc0/input31
[ 1808.204930] usb 1-1.3.4: dvb_usb_v2: schedule remote query interval to 200 msecs
[ 1808.213261] usb 1-1.3.4: dvb_usb_v2: 'Realtek RTL2832U reference design' successfully initialized and connected

Everything appears functional to me.

markwinger, 2021-10-21:
If followed this link to try blacklisting:
https://www.reddit.com/r/RTLSDR/wiki/blacklist_dvb_usb_rtl28xxu

but it did not help.

AlbrechtL, 2021-12-28:
I'm closing this issue because of inactivity.

closed 2021-12-28 by AlbrechtL