AlbrechtL/welle.io #687: Bug scan stations

jjv2021, 2021-11-16:
2021-11-16T20:57:10.987 Info: ofdm-processor: SyncOnPhase failed
2021-11-16T20:57:11.266 Debug: RadioController: Stop channel scan
2021-11-16T20:57:11.267 Info: OFDM-processor: closing down
2021-11-16T20:57:11.273 Info: RTL_SDR: End rtlsdr_read_async_wrapper() thread
2021-11-16T20:57:11.274 Warning: qrc:/QML/MotView.qml:85: TypeError: Cannot read property 'width' of null

Good evening, the scan stops abruptly with the above error and therefore impossible to obtain the list of stations.
jj

nickoe, 2021-12-17:
I get the same on version 2.3 on archlinux.

AlbrechtL, 2021-12-28:
Thanks for reporting.
1. Is it reproducible?
2. Can you try using rtl-tcp -> welle.io via rtl_tcp input?

noone2k, 2022-08-14:
i'm in ... same problem ...
using rtl_tcp works with a little delay ... and sometimes after the second try/channel change ... but once it works, it works ...
btw. latest master-branch build , on arm64 ...

gvanem, 2022-08-14:
The signal level is probably too low. Needs >= 15 dB to get a sync AFAICS.

andimik, 2022-08-14:
No, I don't think this issue is related to the reception.

noone2k, 2022-08-15:
i rebuild the latest master as RELEASE, instead of DEBUG and its working.
a little bit picky but it works ... the 2.2 is not so bitchy.
and yes, where 2.2 lock a transponder/program faster with lower reception,
2.4 lock later and better with a higher snr.
probably a little bit to few overhead with debug or something similar.
i'm running welle-io-gui next to a fully blown gnome desktop on a "weak" sbc ( odroid n2+ )

AlbrechtL, 2022-08-21:
Maybe you run into a unknown race condition because your are using a not really powerful CPU. When building as RELEASE the CPU load is roundabout 50% of the CPU load of DEBUG.

closed 2022-08-21 by AlbrechtL