AlbrechtL/welle.io #696: No more nightly files ?

Observerist, 2021-12-23:
It's been a while ... the (new) nightly builds page hasn't seen any activity since October 2021.

Just hoping, that there is nothing serious behind this "silence".

Where are you Albrecht ? Are you OK ?

andimik, 2021-12-23:
Please close this as this is NOT an issue.

The last commit was in October, therefore there are no later nightly files of course.

https://github.com/AlbrechtL/welle.io/commit/2d7b1c3f68ab92b780357b9844e506376a5e74bc

Observerist, 2021-12-23:
War nur eine besorgte Frage über die Zukunft der Entwicklung. Einen anderen Kommunikationsweg gibt es ja nicht mehr.
Sorry. Closing ....

closed 2021-12-23 by Observerist

andimik, 2021-12-23:
Macht nix.

In case you want to contribute (programming or documentation) please send a (or more) Pull request(s)...

AlbrechtL, 2022-01-06:
> Where are you Albrecht ? Are you OK ?

Thanks for asking. I'm doing pretty well. See my LinkedIn channel: https://www.linkedin.com/in/albrecht-lohoefener/

But I have to say that welle.io is a hoppy project for me which I do just for fun. Thus, I have to split my time with my family (I have 3 small children). Please understand that welle.io has not my top priority. Nevertheless I'm using the program in my daily life (kitchen radio with a Raspberry Pi 4 and official 7" touch display). And from time to time I'm doing coding are maintenance work.

Thanks to the community (especially thanks to @andimik ) I don't have to answer to issues in the first place.