AlbrechtL/welle.io #701: Add button to start station scan

dforsi, 2022-01-06:
Since the first thing after installing well-io is creating the station list, I would find it easier if a button or a link appeared in a prominent place, in addition to the menu item; it could appear also after the user cleared the station list, that is every time the station list is empty, and wouldn't clutter the interface when the list contains at least one station.

A mockup:
[image]

Another mockup:
[image]

AlbrechtL, 2022-01-06:
That's a pretty good point! Thanks!

Pull request are welcome!

AlbrechtL, 2024-10-08:
Added in version 2.5.

closed 2024-10-08 by AlbrechtL