AlbrechtL/welle.io #703: Library version

Observerist, 2022-01-08:
Just found out, there were some new nighties available for the new year. tried out the latest one for Macs, 20220107.

Got this while trying to open that version :

Dyld Error Message:
  Library not loaded: @executable_path/../Frameworks/libfftw3f.3.dylib
  Referenced from: /Applications/welle.io 20220107.app/Contents/MacOS/welle-io
  Reason: Incompatible library version: welle-io requires version 10.0.0 or later, but libfftw3f.3.dylib provides version 9.0.0

Would this be related to a macOS version being too old ?  ( currently High Sierra on a repurposed old Macbook )

Observerist, 2022-01-14:
Is there anything I should have known about this "wrong" version of libfftw3f.3.dylib ? 
I just know that it also happens with macOS 10.14 Mojave. So this library has no (?) connection to the OS. 
Where does this library come from ? Will the next official version of welle.io also use this dependency version ?

AlbrechtL, 2022-02-16:
@orryverducci Can you have a look at this, please?

orryverducci, 2022-02-16:
Unfortunately I can't replicate this issue (on macOS Monterey). The FFTW library included in the 2.4 release is version 10.10.0, which is higher than the required version 10.0.0.

@Observerist could you please try the 2.4 release to see if you are still getting this error.

Observerist, 2022-02-16:
Thanks orryverducci

Where is the 2.4 release ? In the nightlies ?
Would love to have it work on my Mojave mainstay ( even on my High Sierra MBP that's always on my side )

Observerist, 2022-02-16:
Sorry.... forgot to look at the release page.

Will test 2.4 tonight. Thanks again.

Observerist, 2022-02-17:
Same outcome :

Termination Reason:    DYLD, [0x3] Wrong version

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: @executable_path/../Frameworks/libfftw3f.3.dylib
  Referenced from: /Applications/welle.io 2.4.app/Contents/MacOS/welle-io
  Reason: Incompatible library version: welle-io requires version 10.0.0 or later, but libfftw3f.3.dylib provides version 9.0.0

Strange. 
I can see inside of the 2.4 version and the frameworks folder the -libfftw3f.3.dylib-, its version is not revealed.
Could it be that because you built your dmg on Monterey, the FFTW version 10.10.0 can not be implemented when called by a Mojave driven OSX, hence being stuck with 9.0.0 when running under old OSX versions ?

The latest GQRX has the same problem when trying to run the "youngest" app with an "old" OSX.

There was a time, when old computers could still be used for tasks that didn't request the latest powerhouses.
Although that is more relevant with Hams in the Windows/Linux world. 

Was worth a try .... thanks.

orryverducci, 2022-02-17:
There's a few things I think we can look in to here.

I'm wondering if you have another copy of libfftw3f.3.dylib available on your system, and if macOS is loading that instead of the one included in the app bundle.

Could you try opening a Terminal and running the following commands:
```Shell
export DYLD_PRINT_LIBRARIES=1
/Applications/welle.io 2.4.app/Contents/MacOS/welle-io
```

This will launch welle.io with a debug mode enabled that prints the libraries being loaded to the terminal. Could you please copy the output to the terminal and paste it here so we can see what it is trying to load and where from.

> Strange. I can see inside of the 2.4 version and the frameworks folder the -libfftw3f.3.dylib-, its version is not revealed. Could it be that because you built your dmg on Monterey, the FFTW version [...]

I would be surprised if older macOS versions see the version differently. The version of the libfftw3f.3.dylib library can be revealed using otool on the welle-io binary if you have Xcode installed, as you'll see on this screenshot from my Mac:
[image]

If you have otool available it'll be interesting to see what it reports for you. If you don't have it don't worry as it should be the same as what I see.

AlbrechtL, 2022-12-16:
I'm closing it because of inactivity.

closed 2022-12-16 by AlbrechtL