AlbrechtL/welle.io #704: EPG support?

ubuntu-novize, 2022-01-08:
Some of my favorite radio stations transmit program data (like TV).
Do you plan to display this data? EPG = Electronic program guide.

andimik, 2022-01-08:
We already have discussed this at https://github.com/AlbrechtL/welle.io/issues/558

The former EPG is called SPI.

Therefore, you might close this issue as this is a duplicate.

ubuntu-novize, 2022-01-08:
Thank you for this info.
I read #558: to much techno speaking, to much english. Sorry, I don't understand what is discussed there...

andimik, 2022-01-08:
Well, you could give Qt-DAB a try if you want to see what is transmitted in the mux.

Which Mux is it, and where do you live?

AlbrechtL, 2022-02-16:
Duplicated with https://github.com/AlbrechtL/welle.io/issues/558.

Short answer. At the moment there is no EPG support available.

closed 2022-02-16 by AlbrechtL