AlbrechtL/welle.io #705: [Android] Android Auto App Support got lost

pathfinder1803, 2022-01-08:
As for as I understood the Android Auto App Support was integrated before. See Issue #103 
In the current nightly build this feature seems to be lost. Is it because of the issues with the google play store or is it not part of this branch?

Retro64, 2022-02-16:
https://github.com/AlbrechtL/welle.io/pull/674#issue-976431116 mentioned `src/welle-gui/android/res/xml/automotive_app_desc.xml` - the name sounds a bit like Auto related - might this be the point, when it dropped?

AlbrechtL, 2022-02-16:
@pathfinder1803 Yes you are right. The reason is that Google denied to put welle.io with Android Auto support into the Play store. The Google guys wanted to test it but they didn't understand that they need a SDR receiver for it. So they wrote me without a test at Google site there is no approval for the Play store. Furthermore the developer @twoofseven lost his interest in this project. After some time the code was orphaned and not working anymore so I removed it.

closed 2022-02-16 by AlbrechtL