AlbrechtL/welle.io #706: Poor reception and deadlock after changing stations

MightyPork, 2022-01-09:
After starting the app, reception works OK, but when I change channels a couple times, it starts showing errors and stops playing. This time it also ended with a complete lock-up waiting for some futex.

welle.io 2.3 Git: f1ed82e, Arch Linux, rtl-sdr (Nooelec NoSDR stick)

Log: 
[welleio-log.txt](https://github.com/AlbrechtL/welle.io/files/7836155/welleio-log.txt)

trace:
```
Using host libthread_db library "/usr/lib/libthread_db.so.1".
0x00007f59a1f6e8ca in __futex_abstimed_wait_common64 () from /usr/lib/libpthread.so.0
(gdb) bt
#0  0x00007f59a1f6e8ca in __futex_abstimed_wait_common64 () from /usr/lib/libpthread.so.0
#1  0x00007f59a1f63743 in __pthread_clockjoin_ex () from /usr/lib/libpthread.so.0
#2  0x00007f59a0290628 in __gthread_join (__value_ptr=0x0, __threadid=<optimized out>)
    at /build/gcc/src/gcc-build/x86_64-pc-linux-gnu/libstdc++-v3/include/x86_64-pc-linux-gnu/bits/gthr-default.h:669
#3  std::thread::join (this=0x55f1788aa360) at /build/gcc/src/gcc/libstdc++-v3/src/c++11/thread.cc:112
#4  0x000055f176ca6631 in CRTL_SDR::setFrequency(int) ()
#5  0x000055f176c55a21 in CRadioController::setChannel(QString, bool, bool) ()
#6  0x000055f176c5ce7b in CRadioController::play(QString, QString, unsigned int) ()
#7  0x000055f176c3b59d in ?? ()
#8  0x000055f176c409b3 in CRadioController::qt_metacall(QMetaObject::Call, int, void**) ()
#9  0x00007f59a11d5c16 in ?? () from /usr/lib/libQt5Qml.so.5
#10 0x00007f59a10d8f8d in ?? () from /usr/lib/libQt5Qml.so.5
#11 0x00007f59a10d9eb2 in QV4::QObjectMethod::callInternal(QV4::Value const*, QV4::Value const*, int) const () from /usr/lib/libQt5Qml.so.5
#12 0x00007f59a10f9fde in QV4::Runtime::CallPropertyLookup::call(QV4::ExecutionEngine*, QV4::Value const&, unsigned int, QV4::Value*, int) () from /usr/lib/libQt5Qml.so.5
#13 0x00007f593c058255 in ?? ()
#14 0x00007ffeab6b2010 in ?? ()
#15 0x000000000000000d in ?? ()
#16 0x00007f593c058170 in ?? ()
#17 0x0000000000000000 in ?? ()
```

it looks like the binary is stripped, i can try to rebuild it without strip, if it helps.

AlbrechtL, 2022-02-16:
Thanks for reporting! I never experienced your behavior. Which CPU do you have?

MightyPork, 2022-02-16:
I switched to a new computer recently, you should have asked sooner, now I'm not sure which this was on ;)

but probably the new one

```
$ lscpu
Architecture:            x86_64
  CPU op-mode(s):        32-bit, 64-bit
  Address sizes:         43 bits physical, 48 bits virtual
  Byte Order:            Little Endian
CPU(s):                  12
  On-line CPU(s) list:   0-11
Vendor ID:               AuthenticAMD
  Model name:            AMD Ryzen 5 1600 Six-Core Processor
    CPU family:          23
    Model:               8
    Thread(s) per core:  2
    Core(s) per socket:  6
    Socket(s):           1
    Stepping:            2
    Frequency boost:     enabled
    CPU max MHz:         3200.0000
    CPU min MHz:         1550.0000
    BogoMIPS:            6389.26
    Flags:               fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_go
                         od nopl nonstop_tsc cpuid extd_apicid aperfmperf rapl pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 movbe popcnt aes xsave avx f16c rdrand lahf_lm cmp_legacy sv
                         m extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw skinit wdt tce topoext perfctr_core perfctr_nb bpext perfctr_llc mwaitx cpb hw_pstate ssbd ibpb vmmcall
                          fsgsbase bmi1 avx2 smep bmi2 rdseed adx smap clflushopt sha_ni xsaveopt xsavec xgetbv1 xsaves clzero irperf xsaveerptr arat npt lbrv svm_lock nrip_save tsc_scale vm
                         cb_clean flushbyasid decodeassists pausefilter pfthreshold avic v_vmsave_vmload vgif overflow_recov succor smca sme sev sev_es
Virtualization features: 
  Virtualization:        AMD-V
Caches (sum of all):     
  L1d:                   192 KiB (6 instances)
  L1i:                   384 KiB (6 instances)
  L2:                    3 MiB (6 instances)
  L3:                    16 MiB (2 instances)
NUMA:                    
  NUMA node(s):          1
  NUMA node0 CPU(s):     0-11
Vulnerabilities:         
  Itlb multihit:         Not affected
  L1tf:                  Not affected
  Mds:                   Not affected
  Meltdown:              Not affected
  Spec store bypass:     Vulnerable
  Spectre v1:            Vulnerable: __user pointer sanitization and usercopy barriers only; no swapgs barriers
  Spectre v2:            Vulnerable, IBPB: disabled, STIBP: disabled
  Srbds:                 Not affected
  Tsx async abort:       Not affected
```

AlbrechtL, 2022-02-17:
Does it working now?

Most likely you run into a race condition that we don't know. On which CPU did you discover this issue?

MightyPork, 2022-02-17:
Yes, it works now, but two things changed - possibly the CPU, and I also got a better rtl-sdr stick (the rtl-sdr.com blog one).

Here's the old CPU:

```
Architecture:            x86_64
  CPU op-mode(s):        32-bit, 64-bit
  Address sizes:         39 bits physical, 48 bits virtual
  Byte Order:            Little Endian
CPU(s):                  4
  On-line CPU(s) list:   0-3
Vendor ID:               GenuineIntel
  Model name:            Intel(R) Core(TM) i5-6400 CPU @ 2.70GHz
    CPU family:          6
    Model:               94
    Thread(s) per core:  1
    Core(s) per socket:  4
    Socket(s):           1
    Stepping:            3
    CPU max MHz:         3300.0000
    CPU min MHz:         800.0000
    BogoMIPS:            5401.81
    Flags:               fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art
                          arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1
                          sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault invpcid_single ssbd ibrs ibpb stibp tpr_shadow vnmi fl
                         expriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida ara
                         t pln pts hwp hwp_notify hwp_act_window hwp_epp md_clear flush_l1d arch_capabilities
Virtualization features: 
  Virtualization:        VT-x
Caches (sum of all):     
  L1d:                   128 KiB (4 instances)
  L1i:                   128 KiB (4 instances)
  L2:                    1 MiB (4 instances)
  L3:                    6 MiB (1 instance)
NUMA:                    
  NUMA node(s):          1
  NUMA node0 CPU(s):     0-3
Vulnerabilities:         
  Itlb multihit:         KVM: Mitigation: VMX disabled
  L1tf:                  Mitigation; PTE Inversion; VMX vulnerable, SMT disabled
  Mds:                   Vulnerable; SMT disabled
  Meltdown:              Vulnerable
  Spec store bypass:     Vulnerable
  Spectre v1:            Vulnerable: __user pointer sanitization and usercopy barriers only; no swapgs barriers
  Spectre v2:            Vulnerable, IBPB: disabled, STIBP: disabled
  Srbds:                 Vulnerable
  Tsx async abort:       Not affected
```

I'll close this, it works for me fine now

closed 2022-02-17 by MightyPork