AlbrechtL/welle.io #708: no rawfile option under GNU/Linux?

jmfriedt, 2022-01-23:
I am attempting to use a rawfile for decoding DAB in an area where no broadcast is available over the air but the rawfile option does not seem available, neither with a clear compilation from the github repository neither with the Debian/sid version 2.2-1+b1. Running as described at https://www.welle.io/devices/rawfile from the build directory
/tmp/welle.io/build$ ./welle-io -d rawfile --raw-file /tmp/dab_g39_211223.u8 
welle.io: Unknown options: d, raw-file.

tenzap, 2022-01-24:
Did you try to set the raw file in the GUI directly through the input source in the settings?

jmfriedt, 2022-01-24:
Thank you for your reply, I had not realized I could select the source from the GUI. Indeed disabling the Auto Detect feature on Settings allows for selecting RAW file and reading from the archive. 
Maybe the documentation https://www.welle.io/devices/rawfile would deserve an update. 
I can confirm that from there the u8 file is properly interpreted and allows for decoding the whole DAB+ stream.
Thank you.

closed 2022-01-24 by jmfriedt

andimik, 2022-01-24:
Ok, I'll check if I have time to write a Pull Request for https://github.com/AlbrechtL/welle.io/tree/gh-pages (which then updates the webpage). 

But of course you could send a PR, too ...