AlbrechtL/welle.io #710: request to bring well.io to fdroid.org

freddii, 2022-01-31:
It would be nice if welle.io can be added to https://f-droid.org

tenzap, 2022-01-31:
This is not that easy because welle.io is based on Qt. Fdroid build servers use debian, and qt for android is not packaged in debian.
Don't hesitate to contribute to the subjects mentioned below so that they go onward if you would like to see that in fdroid.

For now, according to a comment [here](https://mentors.debian.net/package/qt6/), it is unlikely to see qt-android in debian.

There is another way but much more complex to get that in f-droid by writing a metadata file in fdroid which gets the binaries of the libraries from qt servers. You could do it that way if you want to see this happen.

Few links for those interested:
- https://forum.f-droid.org/t/packaging-qt-apps/14764
- https://gitlab.com/fdroid/fdroiddata/-/merge_requests/9667
- https://mentors.debian.net/package/qt6/
- https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1002004

AlbrechtL, 2022-02-16:
@tenzap Thanks  for clarification. I would be nice to have F-Droid support. But for now it is not our main objective.

This issue is duplicated with https://github.com/AlbrechtL/welle.io/issues/185.

closed 2022-02-16 by AlbrechtL