AlbrechtL/welle.io #712: [welle-cli,feature request] Support more than one channel/ensemble per instance

nRaecheR, 2022-02-11:
Welle-cli features a headless network server mode which is a really nice feature to stream DAB+ over a network and integrate it into other solutions like TVHeadEnd etc. The stream URLs can be downloaded as m3u playlist from a fixed URL.

But it supports only one channel at a time and the channel number needs to be set by command line or web interface. 
It would be nice to support more then one channel/ensemble with help of an initial scan of a list of given channels and then switch the channel/SID on the fly when the stream with a SID has been requested by a http request. Using one SDR receiver per channel/ensemble seems to be overkill when only one service should be played. It should even be possible to support multiple streams per channel/ensemble. That's the way TVHeadEnd is using the muxes on a DVB-T2 or DVB-S2 device.

This would make Welle-cli a full solution as DAB+ streaming server.

AlbrechtL, 2022-02-16:
Great idea! To use welle.io via TVHeadEnd is one if my dream because this makes it possible to integrate it into Kodi and it distributions. The aim for welle-cli is monitoring for DAB broadcasters so it is different from you request.

I saw that you started your own fork of welle-cli https://github.com/nRaecheR/dab_plus_streamer. I invite your to contribute to this project. What is your idea with dab_plus_streamer?

nRaecheR, 2022-02-16:
The goal is to have a streaming server for multiple DAB+ channels as described in the first post here. I learned that I need to do a lot of modifications in the shared code of CLI and GUI, so I think it's better to fork it and remove the QT-GUI completely because I don't want to do work on something I don't need for the core functionality. The same goes for the Windows, Mac and Android support. 
I don't have a lot of spare time so this will be a low prio project for me. I really would like to interface DAB+ to TVHeadEnd with multiple channel and maybe slideshow support.

gvanem, 2022-02-16:
@AlbrechtL What happened to the idea to use SDRDAB as a library in Welle-IO? Ref: https://github.com/AlbrechtL/welle.io/issues/102. I still see you have a [branch](https://github.com/AlbrechtL/welle.io/tree/sdrdab-test) for it.

AlbrechtL, 2022-02-16:
@nRaecheR welle-cli and welle.io GUI are complete independent from each other (different code, different build system, different maintainer, different use case). But both are implementing the interface https://github.com/AlbrechtL/welle.io/blob/master/src/backend/radio-controller.h. I highly recommend that you do the same way. If you do it in the same way you can benefit from the welle.io community and we can benefit from you work (e.g. if you do improvements in the backend). I have no problem to add your implementation as a third application besides welle.io GUI and welle-cli because your use case is complete different.
But anyway you can start as you own. That's open source! If you would like to contribute to welle.io our doors are open.

@gvanem As you see there is no activity since 22 November 2017. The original website https://sdr.kt.agh.edu.pl/sdrdab-decoder/ is down and at https://github.com/kwanty/sdrdab that last commit is from 8 August 2017. So I assume that the sdrdab project is dead. In the last 4 years we improved the backend code a lot so there is no need to change the decoder. Most likely I will remove sdrdab branch.

lo-decibel, 2022-04-08:
> Yep, totally possible. I had DAB radio working in Tvheadend by creating a script to pipe the output of eti-cmdline --> dablin --> ffmpeg --> tvheadend. I just had to create an m3u8 playlist for radio manually. However, I found this approach made my system runstable, and it aldo took about 15 seconds to switch programmes. I'm going to try and find another solution with the welle-cli webserver.

My only gripe with welle is that the webserver transcodes the DAB stream to MP3. Is it possible to output the raw AAC stream and let ffmpeg add the headers for Tvheadend?

rocus, 2022-07-31:
Below a describe a rather hacky way to change the channel of a running welle-cli -w8888 server.  I am not a fan of C (cpp is much better)  and incapable of making changes in a big and complicated program like welle-cli. The result of a very small change in welle-cli and a monitoring script does a channel change as was the purpose. I am not proud of it (everything should happen in welle-cli) but it works.

The change in welle-cli is a few lines that create a file (in /tmp) containing the radio sid that was supposed to play. 
A diff between the old and new webradiointerface.cpp is:

446,447d445
< FILE *fptr;
< 
509,513d506
<                     if ( ! success) {
<                          fptr = fopen ( "/tmp/dabwrongchannel.txt" , "w+" );
<                          fprintf ( fptr , "%s" , req.url.c_str() );
<                          fclose  ( fptr );
<                     }

This is right after the line: success = send_mp3(s, match_mp3[1]);

Together with the starting of welle-cli  there is a script constantly monitoring the existance of the file /tmp/dabwrongchannel.txt.

for (( ; ; ))
do
   if [ -f /tmp/dabwrongchannel.txt ];   then
      sidhex=`sed 's/\/mp3\/0x//' /tmp/dabwrongchannel.txt | tr [:lower:] [:upper:]`
      sid=$(echo "ibase=16; $sidhex" | bc)
      chn=`jq '.[] | select (.stationSId =='$sid' ) | .channelName'  stations.json  | sed s/\"//g`
      echo stationid: $sid stationidhex: $sidhex Newchannel: $chn
      curl -X POST --data "$chn" http://localhost:8888/channel
      rm /tmp/dabwrongchannel.txt
   fi
   sleep  1 
done

If this file exists then a (hex) stationid is extracted. The stationid is converted to decimal and the channel is searched for
this stationid. With curl the currentchannel is changed for the desired channel and the temporary file is deleted. The channelchange takes some time and thereafter the radio station must be selected again on the streamer.
When the playing of a stream goes wrong the first time that is an indication that a channelchange must take place; it should play the second time.

Again, the program should check beforehand if the channel is correct , make, if necessary,  a channel change  and then stream the choosen channel. Maybe somebody is capable of changing welle-cli in such a way.

rocus, 2022-08-02:
Because I found this hack unsatisfactory I decided to try to change the origional welle-cli  program. I made some changes in the program webradiointerface.cpp and very few changes in an other file and got this working. I try to put this all in a pull request. This might take some time because I have not used git before.

AlbrechtL, 2022-08-21:
See PR #748

Lamarqe, 2024-08-06:
I implemented a DAB streaming server which is based on welle.io and handles the scenario discussed above:
https://github.com/Lamarqe/mpdcast-dab