AlbrechtL/welle.io #713: [Gui, Linux] on 2.4, some icons are missing

tenzap, 2022-02-13:
Still the same as here https://github.com/AlbrechtL/welle.io/commit/1c1b7b8bcf8f7792adf5248c040dc2655c4d7e74 with 2.4. I didn't have such a problem with 2.3.

This is with the downloaded appimage on debian bullseye (latest, up-to-date stable). Style= the default one, but all are touched.

AlbrechtL, 2022-02-16:
I can't reproduce your issue. Maybe it is something with your graphic card driver?

tenzap, 2022-02-16:
If it were, icon would be missing in 2.3 also, right? (which is not the case)

tenzap, 2022-02-16:
What about adding both path in that commit?

AlbrechtL, 2022-02-16:
The issue is that I don't understand what the problem is. The new path is the correct once. Why in your case only the wrong path is working I don't have any idea.

andimik, 2022-02-17:
Could you share a screenshot?

tenzap, 2022-02-18:
Screenshot is in https://github.com/AlbrechtL/welle.io/commit/1c1b7b8bcf8f7792adf5248c040dc2655c4d7e74

AlbrechtL, 2022-08-21:
@tenzap PR #723 is merge. Can you confirm the fix, please?

purewasa, 2022-11-15:
Hi @AlbrechtL,

I am running version 2.4 since April 2022 on Rapsberry Pi OS bullseye.
Everything works. Some qml warnings, but no serious problems.
Suddenly, icons were missing. Like in:
https://github.com/AlbrechtL/welle.io/commit/1c1b7b8bcf8f7792adf5248c040dc2655c4d7e74
Must be by an update. (I update once/twice a week.)
Icons were there something before 2022-11-07.
The only major thing I can remeber is a kernel update to:
$ uname -a
Linux raspberrypi 5.15.76-v7+ #1597 SMP Fri Nov 4 12:13:17 GMT 2022 armv7l GNU/Linux

From this issue I tried commit:
https://github.com/dforsi/welle.io/commit/e1784abe8a7fd4f0b35ac15819217276fa25d9b6
to fix the problem, which is in branch next.
Cherry-picking this commit to master of 2022-11-13 (2.4 + 4 commits)
fixes the problem! (I don't have Qt6 installed!)

Kind regards,

purewasa

closed 2022-12-14 by AlbrechtL