AlbrechtL/welle.io #716: Not working on Raspberry Pi 3B+ due to SyncOnPhase failures

lapseofreason, 2022-02-22:
When I try to use welle.io on a Raspberry Pi 3B+ (welle.io_2.2-1_arm64 on Raspbery PI OS 64-bit), I cannot listen to any channel. The console shows frequent messages like this:

`... Info: ofdm-processor: SyncOnPhase failed`

While the program is running, the "Frame sync" is mostly green but often changes to red for a short period of time. The other two status indicators are mostly red, occasionally changing to green for a short period of time.

The CPU load is in the 350+% (Raspberry Pi 3B+ is quad core) while running.

Also when searching for channels, it only finds a couple of them.

I am using a DVB-T stick with antenna with the RTL2832U chipset. The same stick works fine when I use it on my laptop (welle-io-2.3-2.fc35 on Fedora 35). It finds many channels and I can listen to all of them.

Anything I am missing here or is the Raspberry Pi 3B+ just not powerful enough?

andimik, 2022-02-22:
Could you try to record a raw or sdr with your laptop and play this file on the raspberry?

andimik, 2022-02-22:
A *.wav (16 bit) is what some SDR tools describe as *. sdr

For more information on the file format please visit the website.

lapseofreason, 2022-02-22:
So I managed to record a 10 second sample with the dab_raw_record script from [here](https://www.welle.io/devices/rawfile) and played it back using by setting raw file in the GUI.

I could hear some part of the sample when playing it back, but it was only audible for a couple of  seconds (not the full 10 seconds). What does that mean?

andimik, 2022-02-23:
Does it play on the laptop?

Have you tried other software like Qt-DAB or have you compiled welle-cli as well?

I also own a 3B+ but it is working fine, ever with more CPU eating software like eti-cmdline.

lapseofreason, 2022-02-23:
It plays pretty much the same on the laptop, I think the sample might have been too short (10sec).

Yes, I also have welle-cli installed (from repo, not compiled myself), I do not have Qt-DAB installed.

I did some more tests with longer samples (60sec), one recorded on the 3B+ and one one my laptop. I then tried to play them back on the 3B+ and the laptop with welle-cli. There were no differences, both played fine after some initial sync time, I could not notice any differences between the 3B+ and the laptop. It even worked almost the same in the GUI on the 3B+, with one minor skip somewhere in the middle (seemed to be random though).

Still, if I try to listen in live in the GUI it never manages to sync (antenna in the same place as for the pre-recorded sample).

andimik, 2022-02-23:
So I fear you have to find the best gain setting.

lapseofreason, 2022-02-23:
You mean by manually setting the gain? You think this could be the issue?

andimik, 2022-02-23:
Yes, manual gain, not automatic.

And you could check the expert mode and add the constellation diagram.

If possible, please also share a screenshot.

I fear you will see a big, long line in the middle of the spectrum meaning that auto gain does not work here.

lapseofreason, 2022-02-23:
I tried out different gain values, setting them and then waiting for a couple of minutes, but I could not find any that would work.

If I set the value too high, I get "ADC overload" messages. If it is too low, the SNF is very low (which I think is undesirable, is that right?)

Here is a screenshot of the constellation diagram, it changes all the time but is always all over the place like this:

[image]
.

purewasa, 2022-05-01:
Hi @lapseofreason and @andimik,

I guess, I had the same issue on Raspberry Pi 3 with Raspberry Pi OS 32-bit, but I seem to have solved it!

The workaround works for welle.io-2.4 self-compiled on buster, as well as for welle.io debian package on bullseye.

The problem seems to be the CPU load, welle.io produces:

On buster the process runs with about 300% CPU load (having 4 cores) and 14/15 threads according to htop, tree view, counting every line with "welle-io". And after the fix, it runs with about 265% and 13 threads.
Don't use many panels in the GUI or run other programs in paralell, it worsens the time condition! The signal decoding algorithm seems to be very sensitive to delays, caused by running other processes and threads.

I am using welle.io with expert mode and the panels: console output, service overview, MOT slide show, service details, and the station list and it works fine.

I accidentqally had got the chance to be able to launch the program from two different menu programs. The one worked, the other one did not. So, I compared both ways and finally found the solution.

This is what I did:

- Copy the welle-io.desktop file to the user applications directory and edit it:
  ```
  cd ~/.local/share/applications/
  cp /usr/share/applications/welle-io.desktop .
  nano welle-io.desktop
  ```

- Replace the "Exec" key line by the following line 
  ```
  Exec=sh -c "QMLSCENE_DEVICE=software exec welle-io 2>/dev/null"
  ```
- Save it with CTRL-O.
- Start welle.io from the system menu.

Alternatively, you can just copy the "sh" command to the shell.

Here is how it seems to work:
- The trick is the environment variable with this value. It uses the Scene Graph Adaption feature with type "software", which skips all the OpenGL rendering ( https://doc.qt.io/qt-5/qtquick-visualcanvas-adaptations.html )
  This reduces the thread number by one, but also makes the volume control button on welle.io-2.4 vanish (I guess, it can't be rendered any more)
- The redirection to /dev/null turns of logging output (--log-file option did not work!).
  This reduces the load of logs while searching for the right frequency correction and makes the tuning into a station quicker.

That's it!

I hope this works for other.

andimik, 2022-05-01:
Sorry, I meant the spectrum diagram.

AlbrechtL, 2022-08-21:
@purewasa Thanks for the clarification!

closed 2022-08-21 by AlbrechtL