AlbrechtL/welle.io #719: Add info about SDRplay support

AsciiWolf, 2022-03-02:
I am using welle.io with my SDRplay RSPdx on Fedora 35 and it works great, including additional features like antenna port selection. :-)

AlbrechtL, 2022-08-21:
Great!

Feel free to add the support of SDRplay including a how to to our website https://www.welle.io. You can do this via a pull request to the websites repository https://github.com/AlbrechtL/welle.io/tree/gh-pages

AsciiWolf, 2022-08-21:
Hello Albrecht, I will probably wait until #562 gets fixed because now, I have to manually check and uncheck the "Automatic RF gain", then set the "Manual gain" slider every time I start welle.io. This issue may be related to [osmocom/#5562](https://osmocom.org/issues/5562), however that issue was fixed downstream in my Linux distribution (Fedora), but is still present in welle.io.

AlbrechtL, 2022-12-16:
Pull request are always welcome!

laundmo, 2026-02-23:
Note: I'm using SDRPlay RSP1B through SoapySDR directly (no remote) with the following settings:

**Automatic RF Gain**: enabled
**Auto Detect**: enabled

**SoapySDR Settings**
 - **Antenna**: empty
 - **Clock source**: empty
 - **Driver arguments**: `driver=sdrplay` - note, it did not work with the spelling produced by `SoapySDRUtil --probe sdrplay` which is `driver=SDRplay`

Note: It may not work ideally, since the RSP1B has a DAB-Nothch built-in which defaults to enabled, and i dont think theres a way i can disable it through welle.io